OrcaSlicer/OrcaSlicer #9805: Feature request: Rotating hilbert curve infill

Ataraxia-Mechanica, 2025-06-05:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

Hilbert infill is good for parts that require annealing, due to not generating long lines that causes deformation and internal stress when the plastic is heated up. However, since the pattern for every layer is the same, it would worsen parts' rigidity.

[image]

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

For every other layer, rotate the infill by 90 degrees

### Describe alternatives you've considered

Add a modifier every other layer which is super impractical

pi-squared-studio, 2025-06-06:
Maybe you meant 45 degrees, or even 30?

Ataraxia-Mechanica, 2025-06-06:
> Maybe you meant 45 degrees, or even 30?

Honestly just rotating is better, 90 or 45 should all work

pi-squared-studio, 2025-06-06:
Gilbert is an orthogonal filling. A 90-degree rotation will give the smallest bonding effect between the layers.

Ataraxia-Mechanica, 2025-06-06:
> Gilbert is an orthogonal filling. A 90-degree rotation will give the smallest bonding effect between the layers.

Don't know why you are so caught on the angle. as long as there are paths crossing each other on different layers it would work. Don't think you've done actual tests on the strength to have a conclusion.