OrcaSlicer/OrcaSlicer #9813: MacOS Sequoia Printer LAN Connection broken

JDangerPhD, 2025-06-06:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0

### Operating System (OS)

macOS

### OS Version

Sequoia 15.5

### Additional system information

Hardware Overview:

  Model Name:	Mac mini
  Model Identifier:	Mac14,3
  Model Number:	Z16K000R3LL/A
  Chip:	Apple M2
  Total Number of Cores:	8 (4 performance and 4 efficiency)
  Memory:	16 GB
  System Firmware Version:	11881.121.1
  OS Loader Version:	11881.121.1

### Printer

BambuLab X1 Carbon

### How to reproduce

Just upgraded MacOS from Sonoma (printed no problem, as recently as last week) to Sequoia, and now cannot connect to printer. settings are good, keys/access codes are fine, but Orca tries/fails to connect to the printer a half dozen times and then crashes and closes. This occurs whether or not a project is loaded. Have tried uninstall/reinstall, actually tried to downgrade the OS (created Sonoma bootable disk, etc...) but disk volume cannot be downgraded. Grabbed the nightly dev version and installed, same issue.

### Actual results

[image]

### Expected results

the software used to connect to the printer via LAN/access code without issue, but it cannot in this OS.

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [ ] Log file
- [ ] Project file

angel7501, 2025-06-20:
Try checking if your wifi has network bifurcation on. I had the same issue because my printer was connected through my 2.4GHZ network while my laptop was connected to 5G but both displayed the same SSID. As soon as I disabled bifurcation and set up separate networks and connected my laptop to the same 2.4ghz network as the printer, it properly connected and Orca didn't crash again.

Mac OS Sequoia, Orcaslicer 2.3.0.

HexRebuilt, 2025-08-30:
Hi have the same issue, but my mac is connected via ethernet. don't know what to do.  At random it works and i'm able to send prints but it is a 1/50 chances. makes really no sense.

HexRebuilt, 2026-01-06:
same issue. present from months on windows, ubuntu and macos. 
at times eveything  works fine, reliably for days. other times there is no way to connect to the printer via orca slicer. 

Homeassistant works fine, just like the esp32 touchscreens with xtouch. both are able to interact with the printer. HA has no issue streaming as well.