OrcaSlicer/OrcaSlicer #9820: Max Flowrate is Broken

ARand0mGitUser, 2025-06-07:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0

### Operating System (OS)

Windows

### OS Version

11 Pro

### Additional system information

CPU: AMD Ryzen 5 5600X
RAM: 32 GBs
GPU: NVIDIA RTX 2060 KO

### Printer

Prusa Mk3s+ (with klipper)

### How to reproduce

1. Click on the calibration tab
2. Click on more
3. Click Max flowrate
4. Press 'okay' to accept defaults
5. Slice the file
6. Inspect the results

### Actual results

The flow rate of the model is not adjusted until the very top layer, all the rest is the max flowrate, in this case 49.48.

[image]

[image]

[image]

### Expected results

A gradient of flowrate should appear throughout the part, but instead, a thin line appears at the top. 

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/20637645/log.zip)

[project.zip](https://github.com/user-attachments/files/20637650/project.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

### Anything else?

This might just be a fluke on my end, but uninstalling and reinstalling Orca did not change anything. Any help would be appreciated!

Buildasaurus, 2025-06-08:
Trying with the bambu A1 printer, i get a smooth flow rate, following your steps. Switching to the mk3s it's still smooth, but the transition is different. 

[image]

[image]

However i did notice it was important to reclick the "max flowrate" calibration test, and discard all changes, when switching to eg. a new printer or a new filament. Otherwise the calibration would be completely off.

ARand0mGitUser, 2025-06-08:
Strange. There was no variation when I did it, as seen in the screenshots.... i'll try again. Thanks for the reply!

 -ARand0m_GitUser

----------------------------------------

image.png (view on web)
image.png (view on web)

closed 2025-06-08 by ARand0mGitUser

Buildasaurus, 2025-06-09:
Glad you got it working!