OrcaSlicer/OrcaSlicer #9822: Max Flowrate Calibration G-code Generates Incorrect Print Speeds

JoshL93, 2025-06-07:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0 Build 70931e5

### Operating System (OS)

Windows

### OS Version

11

### Printer

Ender 3 V2 with Klipper with 0.4mm Nozzle

### How to reproduce

Use any profile, go to Calibration>More>Max flowrate

Enter any volumetric speed

For example:
[image]

### Actual results

Then you get something like this in Preview:
[image]

F11820 at layer 99 of 100
which translate to print speed of 197mm/s
and it was validated through test print

### Expected results

Based on the Volumetric Flow Rate formula:
Print Speed (mm/s) = Volumetric Flow Rate (mm³/s) / (Line Width (mm) × Layer Height (mm))

With Volumetric Flow Rate 38mm^3/s, the print speed should be around 475mm/s using 0.4mm nozzle with 0.2mm layer height.

so the Gcode generated from this built-in calibration feature is far from the expected value.

Tried with multiple filament/print profiles, and get the same result.

### Project file & Debug log uploads

[Max volumetric speed test.zip](https://github.com/user-attachments/files/20637750/Max.volumetric.speed.test.zip)
You probably don't need this to reproduce, but here it is.

### Checklist of files to include

- [ ] Log file
- [x] Project file

boysie123, 2025-06-08:
whats the light width though? just coz its a .4 nozzle we need to know what the line width is actually to calculate the speed.

JoshL93, 2025-06-08:
Thanks for your comment, I'm new to the software and I'm having a hard time finding the place to set line width, it should be a default value for 0.4 nozzle as I didn't change it. 

> whats the light width though? just coz its a .4 nozzle we need to know what the line width is actually to calculate the speed.

Inmedi14, 2025-06-10:
To calibrate the maximum flow, the programme overrides your object settings. Among others, the layer height and wall width. Check this. In my case, the layer height is 0.32mm and the outer wall width is 0.7mm

Inmedi14, 2025-06-10:
The screen you uploaded is the global settings, click on ‘objects’ above and you will see the changes highlighted in orange.

[image]

JoshL93, 2025-06-11:
> The screen you uploaded is the global settings, click on ‘objects’ above and you will see the changes highlighted in orange.
> 
> [image]

Wow that's why, the test is using a different outer wall width and layer height; the calibration is working as intended after all.

I appreciate your help, closing this ticket now.

closed 2025-06-11 by JoshL93