OrcaSlicer/OrcaSlicer #9827: Rotating Infill

fiveangle, 2025-06-08:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

No

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

It would be awesome to be able to set an infill rotation angle or total number of turns, so that the infill rotates slightly with each layer: It would enable doing things that are currently not directly possible in any slicer that I am aware of:

- Create extra-strong radial structures for improved torque handling capability
- Create amazing new art using any of the static infill patterns printed with no walls
- Allow any of the vertically-static infill patterns to be converted into patterns that generate multiple closed cavities as the pattern rotates through the objects height (kind of the opposite of Gyroid, which is useful in order to ensure a single interconnected air mass within the object)
- Turn any vertical-oriented infill pattern into a fan blade
- Create an Archimedes Screw Pump right in the slicer ;)

But seriously, just trying to visualize the patterns it would create, my mind is reeling with the possiblities.

Ideally, it would have 3 settings that immediately come to mind:

1. Maximum supported rate possible - This would rotate as fast as possible without bridging the infill between inner walls, limited by the rise/run of the extrusion width+layer height at the furthest distance from the center of the infil pattern origin (optionally, perhaps extend this limit by calculating the maximum extrusion width possible as a percentage of the nozzle diameter... a subcheckbox perhaps, else limit to same angle as configured for maximum unsupported overhange angle).
2. Specific angle per layer (something like Arcminutes or Arcseconds per layer ? I can't see a use for radians, but :shrug:)
3. Total number of turns (the infill will rotate NumTurns * (LayersTotal/360) every layer, regardless if doing so will result in any bridging between internal walls)

Anyway, loving OrcaSlicer over Bambu Studio ! Thanks for all the superhuman effort you pour into this wonderful project ♥ 

### Describe alternatives you've considered

Modelling the infill directly in CAD then print without any infill #masochistic

pi-squared-studio, 2025-06-09:
I support this issue. I think this can be easily arranged by adding a numeric parameter instead of this option.
[image]

Here's a similar request, but for a different problem: https://github.com/SoftFever/OrcaSlicer/issues/9805

discip, 2025-09-30:
@fiveangle
What use case do you have in mind where **`1.`** would be needed, other than toying around with infill?😊
As for everything else you asked for, it's already covered.

fiveangle, 2025-09-30:
Rotation any faster will generate infill that is not connected to the outer wall with any appreciable strength. The result will be a step backwards in strength other than potential visual appeal (and I doubt even that). Not having the slicer calculate the optimal #1 in some fashion means that to effectively use this feature for strength or watertight cavities, a user  will need to manually perform their own calculations based on their set layer height, extrusion width, radius from centerpoint to outmost inner wall, and maximum overhang angle. This is tedious trig to do so no one will do it. The only alternative is hitting "Slice", wait, identify the outter-most inner wall visually, zoom in on the UI, adjust the rotation angle manually, completely slicing the entire model between each change, and hopefully iterate enough until there are no gaps and before one dies of old age.

TL;DR: #1 is kind of the magic that makes this feature effective for anything other than artistic use.

Having thought more about it, instead of having a maximum rotation angle, a cleaner interface would be to have the maximum rotation angle populated into the rotation angle field(s) when the "Enable" checkbox for the feature is selected.

fiveangle, 2026-03-18:
sorry i missed this request for clarification. will review, and update or close within a week. thanks !