OrcaSlicer/OrcaSlicer #9874: Window Focus Switches Randomly with Multiple Instances

jamesche75, 2025-06-10:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0

### Operating System (OS)

Windows

### OS Version

Windows 11 24H2

### Additional system information

When multiple instances of Orca Slicer are open and a print is ongoing in at least one of them, the focus randomly switches to another instance while working in an instance. This happens even when dialogs are open that should be modal. For example, when I'm editing a filament profile in one instance, another instance will often activate and focus switches to that one. This can cause keystrokes to unintentionally be entered in the wrong instance. 

I notice this issue when I have two or more instances of Orca Slicer running. 

### Printer

Bambu Lab X1C and Bambu Lab A1

### How to reproduce

1. Launch multiple instances of Orca Slicer. 
2. Start a print in at least one of these instances. 
3. Begin making changes in any instance. This could be changing properties of a model, editing filament profiles, etc.

### Actual results

Frequently, another instance will become the active instance, often in the middle of keystrokes or other changes. 

### Expected results

An instance is only activated when I explicitly activate it. 

### Project file & Debug log uploads

This happens regardless of project. My project files are proprietary to me and are sold in my Etsy shop, so I don't want to include them here.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

jamesche75, 2025-06-18:
Any progress that can be made here would be very appreciated. This just cost me a 15-hour print. I had a VERY long print running on one of my printers. It was 13.5 hours into a 15-hour print. On another printer, I had a print that was just starting (hadn't even loaded filament yet) that I needed to cancel to change a quick setting. I was hovering my mouse over the Stop button in the Device tab and a microsecond before I clicked, Orca switched to the instance running my 15-hour print without me realizing it. When I clicked Stop, it stopped the 15-hour print at 13.5 hours. I now have to start that print all over again, wasting time and filament. I mention this only to illustrate how incredibly frustrating this bug is.

coryrc, 2025-07-29:
The "Failed to connect to the printer" message comes from https://github.com/SoftFever/OrcaSlicer/blob/3f1e4ca55a94dc32000224c20b4dd4533e378383/src/slic3r/GUI/Widgets/SideTools.cpp#L504

Briefly looking, it doesn't appear to steal focus inside there, so something calling `show_status()` with `MONITOR_DISCONNECTED` could be at fault. Also didn't immediately see anything searching.

Looking through wxWidgets docs, either [`RequestUserAttention()`](https://docs.wxwidgets.org/stable/classwx_top_level_window.html#a749d23ea08f6273fb93fe2c003b650b3) (shouldn't be though) or, most likely, [`Raise()`](https://docs.wxwidgets.org/stable/classwx_window.html#a54808c933f22a891c5db646f6209fa4d).

(Note: there's a [`ShowWithoutActivating()`](https://docs.wxwidgets.org/stable/classwx_top_level_window.html#a03e526f505716568318d601318527bd0) method that may-or-may-not help solve this issue.)

There are only two files with calls to Raise() in C++ in this repo, some in ReleaseNote.cpp and one in WebViewDialog.cpp. The latter is suspicious: it's in [`void WebViewPanel::load_url(wxString& url)`](https://github.com/SoftFever/OrcaSlicer/blob/3f1e4ca55a94dc32000224c20b4dd4533e378383/src/slic3r/GUI/WebViewDialog.cpp#L240C1-L240C43).

coryrc, 2025-07-29:
The only WebViewPanel instantiation seems to be here: https://github.com/SoftFever/OrcaSlicer/blob/3f1e4ca55a94dc32000224c20b4dd4533e378383/src/slic3r/GUI/MainFrame.cpp#L1075

Whenever it receives a `EVT_LOAD_URL` event, it calls load_url() which calls Raise(). But... it's not a top-level window AFAICT:

Raise() says `This function only works for wxTopLevelWindow-derived classes` and it inherits from wxPanel. So it *shouldn't* be the cause.

coryrc, 2025-08-09:
I just realized, after following the discussion about Device tab causing slowdowns, that it very well could be the loaded web page triggering a raise. So I (or someone else) might be better off tracing back from the lowest-level user library talking to the window system (i.e. I think wxWidgets uses GTK uses Xlib on Linux) which raises windows. I've never used Xlib so this'll be a while.

coryrc, 2025-08-09:
Ah, well, it might be pretty simple: https://tronche.com/gui/x/xlib/window/XRaiseWindow.html

coryrc, 2025-08-09:
Oh, other tip, before running gdb I did `export DEBUGINFOD_URLS="https://debuginfod.debian.net"` which makes it automatically download symbols for all the used OS packages!

coryrc, 2025-08-09:
In the backtrace, we have `gdk_window_show_internal`; I assume closely related to `gdk_window_show` which is documented as [Like gdk_window_show_unraised(), but also raises the window to the top of the window stack](https://docs.gtk.org/gdk3/method.Window.show.html). So problem is further up.

coryrc, 2025-08-09:
Well, I think see why it's happening multiple times [on Linux](https://github.com/SoftFever/OrcaSlicer/blob/6f44ca576ce5144f8c367efcacfb5080e667e68c/src/slic3r/GUI/Tab.cpp#L480), but that's not affecting the creator of this bug, for he uses Windows.

```
#ifdef __linux__
        // Events queue is opposite On Linux. wxEVT_SET_FOCUS invokes after wxEVT_TAB_SEL_CHANGED,
        // and a result wxEVT_KILL_FOCUS doesn't invoke for the TextCtrls.
        // So, call SetFocus explicitly for this control before changing of the selection
        m_tabctrl->SetFocus();
#endif
```

coryrc, 2025-08-09:
In [`src/slic3r/GUI/Tab.cpp:480`](https://github.com/SoftFever/OrcaSlicer/blob/6f44ca576ce5144f8c367efcacfb5080e667e68c/src/slic3r/GUI/Tab.cpp#L480), the source of this raise is `SetFocus()` on a class descending from wxWindow. I'm filing a bug on wxWidgets, but this particular one might not bet the cause of the later raises (will see in future investigation).

coryrc, 2025-08-09:
I filed the bug on wxWidgets, but this particular issue is probably only affecting Linux unless Orca is using wxGTK on Windows, which is rather unlikely.

So I'll need to keep going and start a print so I can print the call chain for those raises.

SoftFever, 2025-08-11:
When the focus switches, do you notice anything in that Orca instance, such as a modal dialog pop-up or a notification appearing?

coryrc, 2025-08-11:
@SoftFever While this issue is labeled "with Multiple Instances" there's plenty of focus stealing with just a single instance running. I will sync to head though and continue testing.

coryrc, 2025-08-11:
> When the focus switches, do you notice anything in that Orca instance, such as a modal dialog pop-up or a notification appearing?

I get it most often in the early stages of the print, as it transitions through nozzle cleaning and vibration and waiting for heatbed to warm. This happens with single instance and I'm just trying to do other stuff on the computer.

SoftFever, 2025-08-13:
> [@SoftFever](https://github.com/SoftFever) While this issue is labeled "with Multiple Instances" there's plenty of focus stealing with just a single instance running. I don't even think the multiple instances matters, I think it's more just having focus elsewhere. I will sync to head though and continue testing.

#10374 should fix both scenarios. After the fix, Orca will still attempt to grab focus if there is a modal dialog or a block notification; in other cases it should not.

coryrc, 2025-08-14:
It's not raising after slicing a plate; I think it was before.

It's still raising the main frame multiple times on startup (when frame is first raised, and after/same time as Home tab actually loads. Have not started a print yet.

jamesche75, 2025-08-30:
Sorry about taking so long to get back here. I will test the latest build and see if it fixes it. Thank you very much!

SShadow1974, 2025-09-10:
Still an issue with the nightly build as of 8 days ago. This stealing of focus only happens when I'm in another Orca instance and is repeatable consistantly whenever the printing instance changes filament. It is quite annoying to be working in another instance only to have the printing instance switch itself to the active window over and over again. And as others have mentioned, it only happens with the device tab selected on the printing instance.

coryrc, 2025-11-06:
The last call within Orca appears to be wxSimpleBook::SetSelection(), which says:

```
Notice that the call to this function generates the page changing events, use the [ChangeSelection()](https://docs.wxwidgets.org/stable/classwx_book_ctrl_base.html#ab938512b16be50849c1c4888ffa6aa29) function if you don't want these events to be generated.
```

So that might be an option.

This might also be a Linux-only issue due to https://github.com/wxWidgets/wxWidgets/issues/25691

@Dinth if you get a chance, could you try a recent nightly version and see if the multiple-focus-stealing during the start of printing are gone on your system?

coryrc, 2025-11-06:
> Still an issue with the nightly build as of 8 days ago. This stealing of focus only happens when I'm in another Orca instance and is repeatable consistantly whenever the printing instance changes filament.

@SShadow1974 what OS are you using? EDIT: I'm guessing Windows 11.

Perhaps what I called out in the previous comment is affecting Windows too.

@jamesche75 I assume you are using AMS?

coryrc, 2025-11-06:
Also, I realized that since we're vendoring in wxWidgets, I could just fix the bug in wxWidgets too.

31239899, 2026-10-30:
Still happening with 2.4.2. I'll check again with 2.5.0 after it's out of beta.

31239899, 2026-11-06:
It seems like maybe @jamesche75 has to comment to keep this open, since my comment didn't trigger removal of the `stale` tag.