OrcaSlicer/OrcaSlicer #9875: OrcaSlicer crashes trying to open a large file (big model)

plandevida, 2025-06-10:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0

### Operating System (OS)

macOS

### OS Version

macOS Sequoia 15.5

### Additional system information

  Model Name: MacBook Air
  Model Identifier: Mac14,2
  Chip: Apple M2
  Total Number of Cores: 8 (4 performance and 4 efficiency)
  Memory: 8 GB

Apple M2:

  Chipset Model:	Apple M2
  Type:	GPU
  Bus:	Built-In
  Total Number of Cores:	8
  Vendor:	Apple (0x106b)
  Metal Support:	Metal 3
  Displays:
BenQ PD3200U:
  Resolution:	6016 x 3384
  UI Looks like:	3008 x 1692 @ 60.00Hz
  Main Display:	Yes
  Mirror:	Off
  Online:	Yes
  Rotation:	Supported

### Printer

Prusa Mini+

### How to reproduce

1. Open OracaSlicer 2.3.0
2. Drop the file attached onto the UI 
[Frame double Z guide.step.zip](https://github.com/user-attachments/files/20673163/Frame.double.Z.guide.step.zip)
3. The loading model dialog open and start loading
4. Crashes and closes suddenly

### Actual results

The application suddenly closes.

macOS cash report below:

[macOs crash report.pdf](https://github.com/user-attachments/files/20673322/macOs.crash.report.pdf)

### Expected results

I expected the model to load and see it render on the build plate or OrcaSlicer to show an error if it couldn't handle the file size or anything.

### Project file & Debug log uploads

[Frame double Z guide.step.zip](https://github.com/user-attachments/files/20673236/Frame.double.Z.guide.step.zip)

No project file available since the file never loaded.

[log.zip](https://github.com/user-attachments/files/20673264/log.zip)

### Checklist of files to include

- [x] Log file
- [ ] Project file

tlhintoq, 2025-06-12:
You're not really going to try to print that whole as one piece like that - are you?

[image]

Loaded on my windows machine.  How powerful is your Mac?  maybe just not enough RAM to do the job?

[image]

Did you get the import dialog?  I *did* tick the option to split the compound into multiple objects.

[image]

tlhintoq, 2025-06-13:
8gig of RAM?  That's not much. By today's standards and OSes that's a secretarial machine.  I'm running 64 and it took a couple minutes to process that file.

plandevida, 2025-06-13:
No no. It just crashed and close.

8GB it’s not too much but I still do CAD, web development and some docker stuff. So if I am able to do the CAD and more Orca should be able to load the file, maybe it uses more memory for other reasons. But at least it should not crash.

closed 2025-09-19 by github-actions[bot]