OrcaSlicer/OrcaSlicer #9892: Cannot open sublinks in the Calibration wiki

snailcatcher, 2025-06-12:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

doesnt matter

### Operating System (OS)

Linux, macOS, Windows

### OS Version

doesnt matter

### Printer

doesnt matter

### How to reproduce

Go to the [calibration wiki](https://github.com/SoftFever/OrcaSlicer/wiki/Calibration) and click one of the headings for the calibration methods. If I do so I will land at [the homepage of the wiki](https://github.com/SoftFever/OrcaSlicer/wiki/) and following is shown to me:
"You do not have permission to update this wiki."
[image]

### Actual results

doesnt matter

### Expected results

I want to see the calibration wiki site for individual methods.

### Project file & Debug log uploads

doesnt matter

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

Great project btw!

moth4017, 2025-06-12:
same issue when in the device tab , i cannot access external links

[image]

Noisyfox, 2025-06-13:
@ianalexis could you take a look? I think you might missed some of the fix to removing the `.md` suffix from the links in #9858.

ianalexis, 2025-06-13:
> same issue when in the device tab , i cannot access external links
> 
> [image]

i dont get this, my printer "device" tab dosnt show nothing like that.