OrcaSlicer/OrcaSlicer #9899: Slicer Generates Overlapping Paths

pm0u, 2025-06-13:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0

### Operating System (OS)

Linux

### OS Version

Manjaro, Up to date

### Additional system information

```
$ inxi
CPU: 16-core (6-mt/10-st) Intel Core Ultra 7 155H (-MST AMCP-)
speed/min/max: 400/400/4400:4700:3800:2500 MHz Kernel: 6.6.90-1-MANJARO x86_64
Up: 5d 1h 8m Mem: 27.8/62.27 GiB (44.6%) Storage: 931.51 GiB (7.0% used)
Procs: 492 Shell: Zsh inxi: 3.3.38
```

### Printer

Qidi Plus 4

### How to reproduce

Slice the "Filter body" plate in the project provided
Slice the "Filter Body single" plate in the project provided

### Actual results

"Filter body" -> Look at layer 85 and see that the solid infill overlaps all walls

[image]

"Filter Body single" -> Look at layer 86 and see that one of the filter bodies does not have a top layer infill but it does have a single wall

[image]

### Expected results

"Filter Body" would just have solid infill that filled the space between the walls
"Filter Body Single" would have a top layer infill

### Project file & Debug log uploads

[proj.zip](https://github.com/user-attachments/files/20733149/proj.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

### Anything else?

I have tried tweaking a lot of settings and nothing seems to change it - min shell thickness was my main suspicion but this didn't seem to change anything. Infill/wall overlap either.

pm0u, 2025-06-13:
The print of course did not turn out well, I didn't examine the gcode paths before printing this one

[image]

pm0u, 2025-06-13:
I also tried importing into Blender to use the "Make Manifold" option of the 3d printing toolkit extension and the issue is still present.

The model can be found here (I found the issue present on the GH model as well as printables):
https://github.com/nateb16/VoronUsers/blob/master/printer_mods/nateb16/THE_FILTER/THE_FILTER_STANDARD_BED_SPACING/STL/filter%20housing.stl
https://www.printables.com/model/334276-the-filter-for-voron-24

pm0u, 2025-06-13:
I just tested this quickly across all the AppImage versions that are available under releases, v2.1.1 was the first official release going backwards that did not have the issue

MxBrnr, 2025-06-14:
> Interesting! thanks for pointing that out. Some parts of this are inherited from a profile I was provided by someone else, can't say for certain why those were adjusted.
> 
> I definitely prefer that fix to any of the other ones I found. I suppose that could make this solved?

Personally, this is why I never trust profiles from other people, and why I advise others not to either. You will spend less time and frustration taking a default profile and tuning it yourself, than trying to figure out why someone else's profile isn't working properly for you.

Whether the issue is considered solved or not is up to you. I can't see it needing anything else.

pm0u, 2025-06-14:
Confirmed this does slice correctly now, thank you.

I guess I am a bit torn about whether the issue would be solved - I'm going to mark this closed as my particular issue is addressed but some part of me feels that there shouldn't be a possible settings arrangement that produces gcode like this - not sure if that is something that would be too complex/out of scope, against the goals of this project, etc so not trying to push on that just my thoughts as a user.

closed 2025-06-14 by pm0u

pm0u, 2025-06-15:
Just for the sake of completeness, not sure if this would mean its more or less likely to be an actual bug... a "Slice gap closing radius" of 0.026 was the only value I can find that caused this issue. 0.025 and 0.027 and any other random values I tried are fine