OrcaSlicer/OrcaSlicer #9903: Add function: "Avoid VFA speed range".

AlatusAles, 2025-06-14:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

Adding a function to automatically avoid speed ranges where vertical artifacts (VFA) appear.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Translated into English by Machine...

Hello. I think this function can be useful for outer and maybe inner perimeters.

Example (all numbers are not critical):
Slicing a model with a layer height of 0.2 mm and setting the speed of the outer perimeters to 120 mm/sec - everything is ok, but if I increase the layer height to 0.3 mm, then I am limited by the maximum volume flow rate of my hotend and the speed automatically decreases to 90 mm/sec, and at this speed -  the VFA peak of my printer.

Yes, I can check the speed after slicing, but it takes time and any user can forget to do it...

Something: Add to the printer settings, on the "Motion ability" tab, a field "Avoid speed ranges" where you can enter numbers like "20-40, 80-95, 150-180". Add a difference for the minimum speed (also defined by the user, for example 5 mm/sec). It might be useful to set the rule "Only for outer perimeters". When slicing, if the speed falls into these ranges, then reduce the speed below the minimum in the specified range minus the difference (Speed ​​88 = Bad! -> 80 minus 5 = 75 = Good!). After slicing, you can show a yellow banner like: "VFA speed detected. Speed ​​reduced.".

I am not a programmer and cannot help with coding, but I can test.

Noisyfox, 2025-06-15:
https://github.com/SoftFever/OrcaSlicer/pull/9403 Does this fit your need?

AlatusAles, 2025-06-15:
> [#9403](https://github.com/SoftFever/OrcaSlicer/pull/9403) Это соответствует вашим потребностям?

Yes, I believe this is it. Sorry for not being able to find this development thread... You can close this issue, considering that similar functionality is already in testing, or leave it open for ideas, or merge it with existing issues. Thanks!

closed 2025-06-16 by Noisyfox