OrcaSlicer/OrcaSlicer #9938: Cannot print, missing word: "Printing high temperature with " " + "may case damage. Please fix [...] cannot continue"

inakarmacoma, 2025-06-18:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.1-dev

### Operating System (OS)

Windows

### OS Version

Windows 11 Pro

### Additional system information

12th Gen i7-12700H, 64 GB RAM, 3070 Ti

### Printer

Bambu Labs X1 Carbon

### How to reproduce

Steps to reproduce (current stable and latest nightly)
- New project
- Calibration>Flow rate>YOLO
- Verify printer is connected via LAN-only mode
- Verify AMS slot 3 is PETG
- Verify calibration model is set to Slot 3 & PETG
- Verify PETG profile temperature are within nominal ranges
- Verify printer is set to hardened steel nozzle
- Slice
- Print plate
- Select PETG
- select 'bed-leveling' and 'enable ams'
- de-select 'flow dynamics calibration' and 'timelapse'
- 'send'

### Actual results

Observed behavior
- Error message (see screenshot) // cannot continue. 

![Image](https://github.com/user-attachments/assets/a065c0f6-d5ae-4809-a850-37123e

[image]

a79452)

[image]

### Expected results

Expected behavior:
- Calibration sent to LAN-only print
- Normal printing behavior for calibration

### Project file & Debug log uploads

### Checklist of files to include

- [x] Log file
- [x] Project file

### Anything else?

I noticed I'd not set my filament to PETG directly on the printer; I did that, and the error message was still present (see second screenshots fo variations)

From Debug log:
- SupportSpotsGenerator seems like a possible culprit e.g. if PLA was an interface layer or similar. But this is PETG only and supports are disabled.
- Unsure why it's trying to access F:\ which would be the SD card if it were plugged in; and I'm not trying to print a wallet so? Heh...

StarTakko, 2025-06-19:
Following, I have the same error 😢

kmid, 2025-06-22:
I get the same issue. Running Orca Slicer version 2.3.0, with X1 Carbon in LAN mode and firmware 01.09.01.00. 

As soon as I upgraded my firmware from 01.08.02.00 to 01.09.01.00 and switched to LAN mode I started getting the error message described above in Orca when I try to print anything.

TwanVooijs, 2025-06-23:
I had the same issue, solved it by downgrading the firmware one version

kmid, 2025-06-24:
> I had the same issue, 'solved' it by downgrading the firmware one version

Unfortunately 01.09.00.00 has a couple of bugs  that make it quite annoying to use if you have multi AMS

goofballtech, 2025-06-26:
Another of the same on X1C in LAN mode on both PLA and PETG.

CedricE11, 2025-06-28:
I'm also experiencing this. I'm using a X1C printer in LAN mode and I had the error with both PLA and PC.

goofballtech, 2025-09-10:
> Is there a fix to this? Just tried to print on my x1c first time in months and got this same error.

I am running the new alpha release and printing to my X1C fine. So its fixed in the newer versions. Got some other cool features in there if you are willing to give it a shot. Been running smooth so far, not buggy or anything in my expierence.