OrcaSlicer/OrcaSlicer #9939: Batch Pressure Advance Calibration changes heatbed temperature

JannKleen, 2025-06-18:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0

### Operating System (OS)

macOS

### OS Version

15.5

### Printer

Voron Trident

### How to reproduce

1. Start printing Batch PA Calibration
2. Wait for second layer

### Actual results

Heatbed temperature changes from 110 to 60

### Expected results

Heatbed temperature stays the same. 60 is my fallback temperature in the PRINT_START Macro:

```
    {% set BED_TEMP = params.BED_TEMP|default(60)|float %}
```

Which is set in the Machine start G-Code:
```
PRINT_START EXTRUDER_TEMP=[nozzle_temperature_initial_layer] BED_TEMP=[hot_plate_temp_initial_layer] CHAMBER_TEMP=[chamber_temperature]
```

### Project file & Debug log uploads

[Pressure Advance Test.3mf.zip](https://github.com/user-attachments/files/20803983/Pressure.Advance.Test.3mf.zip)

[debug_Wed_Jun_18_20_10_23_27252.log.0.zip](https://github.com/user-attachments/files/20803994/debug_Wed_Jun_18_20_10_23_27252.log.0.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

### Anything else?

I'm not sure whether this is classifiable as a bug, but if not, maybe mention the potential issue when creating the gcode, since this never happens with 'normal' prints.

Noisyfox, 2025-06-20:
Could you also upload the gcode file? Thanks.

Noisyfox, 2025-06-20:
[image]

Noisyfox, 2025-06-20:
[image]
Make sure you selected the correct plate type.

Noisyfox, 2025-06-20:
> ```
> PRINT_START EXTRUDER_TEMP=[nozzle_temperature_initial_layer] BED_TEMP=[hot_plate_temp_initial_layer] CHAMBER_TEMP=[chamber_temperature]
> ```

You hardcoded to use the temp of the hot plate for your first layer, but you selected textured pei plate so it uses the pei temp for other layers.

closed 2025-09-27 by github-actions[bot]