OrcaSlicer/OrcaSlicer #9940: Load modifier from step-file fails

Grumium, 2025-06-18:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.1-dev Build 5041e07

### Operating System (OS)

Windows

### OS Version

Windows 10

### Printer

does not matter

### How to reproduce

Right click on model, add modifier, load...
[image]

select arbitrary STEP file. Yield:

[image]

It works with .3mf files.

### Actual results

Modifier is not added.

### Expected results

Modifier appears.

### Project file & Debug log uploads

the debug log contains the following error:

[error]	2025-06-18 23:33:08.782521[Thread 0x00005a04]:Slic3r::GUI::Tab::tree_sel_change_delayedcan not find page with current selection 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Noisyfox, 2025-06-19:
Please try #9946

Grumium, 2025-06-22:
> Please try [#9946](https://github.com/SoftFever/OrcaSlicer/pull/9946)

Thank you for making the effort. I'm sure it's fixed. A new nightly build would be helpful and convenient for me to confirm.

closed 2025-06-23 by Noisyfox