OrcaSlicer/OrcaSlicer #9941: Keep Solid Infill Angle Aligned with Model Rotation (Not Build Plate XY)

WildFrogg, 2025-06-18:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

Yes — when I rotate a model on the build plate (e.g. 45° to fit larger parts), the solid infill direction doesn’t rotate with it. This causes infill to be misaligned with the part’s geometry, reducing print strength and quality. Currently, I have to manually calculate and apply infill direction per part, which is time-consuming and inefficient.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

🧩 Feature Request: Keep Solid Infill Angle Aligned with Model Rotation (Not Build Plate XY)

When a part is rotated on the build plate (e.g., 45° to fit a larger model), the solid infill direction remains fixed (e.g., 45° globally), causing it to become misaligned with the rotated outer walls. This results in inconsistent strength and poor print quality.

✅ Feature Proposal

Automatically adjust solid_infill_direction based on the part’s XY rotation, so the infill maintains a consistent angle relative to the model, not the build plate.

✅ Benefits

    Improved strength by aligning infill direction with the actual model geometry.

    Cleaner solid surfaces by keeping infill aligned with outer walls.

    Zero manual setup — no need to calculate and manually enter angles.

    Better user experience for anyone rotating parts to fit limited build volumes.

⚙️ Possible Implementation Ideas

    Auto-detect XY rotation from the model and apply solid_infill_direction automatically.

    Could be implemented as part of the slicing engine or through a dynamic modifier.

    Optional toggle setting:
    sync_solid_infill_with_model_rotation = true

🧪 Example Scenario

    You rotate a rectangular box by 45° to fit diagonally on the bed.

    🟥 Current behavior: Infill stays at 45°, now misaligned with part walls.

    ✅ Proposed behavior: Infill also rotates with the part — maintaining optimal alignment.

### Describe alternatives you've considered

I spent several hours trying to solve this with my own script/plugin — including working with ChatGPT to analyze .3mf and .stl files and auto-adjust infill angles based on model rotation. Despite our best efforts, it proved too complex to implement reliably from the outside due to how OrcaSlicer handles transforms.

That’s why I’m submitting this as a feature request — because I believe this functionality belongs at the slicer level, where it can be implemented cleanly and effectively by the OrcaSlicer developers.

WildFrogg, 2025-06-18:
Here’s a visual example:
Rotated part at 45°, but solid infill direction remains fixed, causing misalignment.

[image]

WildFrogg, 2025-06-19:
as far as i can see at #9890  Solid infill direction still stays RELATIVE to BUILD PLATE X Y and not model orientation so u will still need manually adjust your angle if u rotate part on build plate . What im trying to to is option where u check box and if u have infill set at 45deg and ur part is aligned lets say with X axis but u get error message that says that part is too big and u rotate it at 45deg so it can fit on build plate SOLID infill needs to rotate as well to stay at 45deg and as u can see on screenshot that's not happening.

I dont like AI as well i know a lot of people hate when u use it and post about it but i have no clue how to code or make plugins thats why i went with GPT route i did all i can to try make this on my own nothing worked

pi-squared-studio, 2025-06-19:
> #9890

Here is the new location of this project https://github.com/SoftFever/OrcaSlicer/pull/9954

pi-squared-studio, 2025-06-19:
> Solid infill direction still stays RELATIVE to BUILD PLATE X Y

Try the latest update. Rotation should be applied for all infills (except adaptives), unless, of course, the associated information about the rotation of the model is preserved. There is still a moment when such an angle is calculated in the algorithm of a certain fill, but this should be considered as a separate case.

[image]
[image]

WildFrogg, 2025-06-19:
how can i test this in my OrcaSlicer i have last version i think that this is just what i asked for i dont care about sparse infill but i do care about SOLID infill rotation and TOP surface

WildFrogg, 2025-06-20:
that's too mutch work for someone who has no clue about coding and building programs i will have to wait until devs include that option and i hope its gonna be soon

closed 2025-06-22 by WildFrogg