OrcaSlicer/OrcaSlicer #9943: Orca doesnt detect kde dark theme

Prazgreen1, 2025-06-19:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0

### Operating System (OS)

Linux

### OS Version

Fedora 42 (with KDE Plasma DE)

### Printer

-

### How to reproduce

1. Set global theme to Breeze Dark in KDE Plasma
2. Open Orca Slicer

### Actual results

The app is in light mode
[image]

### Expected results

The app should be in dark theme

### Project file & Debug log uploads

debug logs are useless in this case since it appears as though OrcaSlier cannot recognize the theme set in KDE Plasma

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Noisyfox, 2025-06-19:
Please try latest nightly build.

Prazgreen1, 2025-08-17:
Actually, I think it was really flatpak issue. I would love if we were able to get a light/dark theme switcher in orca for Linux as well as mac as it is quite useful.