OrcaSlicer/OrcaSlicer #9969: gcode visualizer

pathurus, 2025-06-22:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

when i'm slicing something, sometimes i'm looking at the layers and I think to myself...

Why can't orca slicer just play the gcode from the start like cura?

so yeah, pls add this

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Make it so that orca slicer can play the gcode from bottom to top

jessy2027, 2025-07-03:
Hi @pathurus, can you give me more details please

pathurus, 2025-07-04:
here:

https://github.com/user-attachments/assets/e7376c50-06f6-494f-9e9d-f6a6c849bd65

jessy2027, 2025-07-04:
Hi @pathurus, I will try to reproduce that. I promise nothing but I will try