OrcaSlicer/OrcaSlicer #9978: Blank screen on Preview & Prepare page

fbp172924, 2025-06-23:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0 70931e5 (Flatpak)

### Operating System (OS)

Linux

### OS Version

Arch Linux 6.15.3-arch1-1 x86_64

### Additional system information

CPU: AMD Ryzen 5 7600X (12) @ 5.46 GHz
GPU: NVIDIA GeForce RTX 4060 Ti 16GB [Discrete]
31GB Ram

### Printer

Bambu Lab A1

### How to reproduce

Launch orca slicer flatpak
Go to preview tab

### Actual results

[image]

[image]

### Expected results

Show the build plate

### Project file & Debug log uploads

No project file

### Checklist of files to include

- [ ] Log file
- [ ] Project file

zbaty1, 2025-06-23:
A previous thread said to add
``` __EGL_VENDOR_LIBRARY_FILENAMES=/usr/share/glvnd/egl_vendor.d/50_mesa.json```
to the flatpak env variables using flatseal. This did not work for me (I am still having this issue as well) but it might work for you
https://github.com/SoftFever/OrcaSlicer/issues/6900#issuecomment-2476982025

closed 2025-06-24 by fbp172924

donslice, 2025-07-31:
Experiencing this issue myself. EndeavourOS 6.15.8-arch1-1

Setting `GBM_BACKEND=dri` worked, but made the viewport very laggy.