OrcaSlicer/OrcaSlicer #9980: Multiple printers on the same profile

SkyTheCodeMaster, 2025-06-23:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

I have multiple of some printers, like 3 Elegoo Neptune 4 Pro, 2 Elegoo Centauri Carbon, etc
Right now, each printer has its own entry in my printer list, like CC-01, and CC-02, and sometimes it can be a little confusing which printer I'm sending to, and when I switch the printer to a different one (but still the same machine), I have to reslice the print even when it generates perfectly identical g-code.

### Which printers will be beneficial to this feature?

Klipper

### Describe the solution you'd like

I would like something similar to BambuLab's print dialog, where I can choose the printer there, and only have one profile for each in my printer list. It would make it significantly easier to know which printer I'm sending to as I'm not relying on the cached printer from other OrcaSlicer instances, and instead it is directly in front of me when I'm sending the print job.

As for each machine being slightly different, that should be handled in the firmware, or just lock this setting behind a "I know what I'm doing" checkbox in preferences.

I have already tried enabling the "Multi Device Management" checkbox in preferences, but it seems to be a BambuLab thing, and not applicable for all printers.

Felix14-v2, 2026-08-31:
Assigning multiple IPs to a single profile and adding a basic routing window would be really handy for people with two or more devices of the same model. For example, K1 SE, which  is relatively cheap and supports WiFi connection. I have 8 of them in my classroom, and _in theory_, I could just hit the `upload and print` button and select between available printers, but in practice I have to travel between them with a stick like an australopithecus

Felix14-v2, 2026-09-01:
The interesting part is that Orca actually supports basic routing: you need to set the same hostname in each printer firmware (like `k1se.local`) and bind a static row of IPs in your router (e.g. `192.168.0.101-108`). As soon as `k1se.local` is added to the printer profile, this will allow you to trigger the IP resolver by pressing the `Upload` button, and if there are multiple ones, Orca will show you a selection window.

But this requires low-level configuration and some basic Linux knowledge, so that's not a solution for everyone. Also this approach doesn't check whether the selected printer is busy, and that can lead to problems.

vndk36, 2026-10-01:
Also interested. I mean the bambu dialog thet I have when sending a print is alread included in orca and as other pointed out, the this was already modifed and is working for creality printers as well.