OrcaSlicer/OrcaSlicer #10000: size bug when rotates object

gabriellimafoto, 2025-06-24:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0

### Operating System (OS)

Windows

### OS Version

win 11

### Additional system information

Hello, how are you? I would like to report a problem with ORCA. When we place an object on the table and check its scale, it is correct. However, if we rotate the object and check the scale again, it will be modified in relation to the colors of the corresponding axes.

### Printer

neptune 4 max

### How to reproduce

1 put a object (rectangle) in the plate;
2 verify the scale (1 color) and write it on a paper to remember;
3 rotate 90 degrees the object 
4 verify scale again (now is diferent)

### Actual results

size measurements are different when rotated

### Expected results

size measurements are different when rotated

### Project file & Debug log uploads

[Desktop.zip](https://github.com/user-attachments/files/20890393/Desktop.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

kisslorand, 2025-06-25:
It is basic geometry. Scale represents the size of bounding box of that particular object. As you rotate an object its X, Y and Z axis dimension are changing. Just think of a stick's shadow as you rotate it, the shadow might change it's length even thou the stick remains the same.

Noisyfox, 2025-06-25:
Try the latest nightly build, which provide ability to switch between different coordinate systems.