OrcaSlicer/OrcaSlicer #10002: [License Inquiry] Is OrcaSlicer allowed for commercial use?

Louis-3dm, 2025-06-25:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

This is not related to a software problem. I would like to confirm whether OrcaSlicer is allowed to be used for commercial purposes.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

I would like to confirm whether OrcaSlicer can be used for commercial purposes (e.g. in business use, product manufacturing, or professional services). Please let me know if there are any restrictions or licenses required.

Noisyfox, 2025-06-25:
I believe you could use it for anything as long as your bussiness does not violate GNU Affero General Public License v3.0.

discip, 2025-06-26:
@Louis-3dm
As this seems to be solved, please don't forget to close. 😊