OrcaSlicer/OrcaSlicer #10029: Memory Access Violation when attempting to open preferences

xenia-foxtrot, 2025-06-29:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0

### Operating System (OS)

Linux

### OS Version

Arch Linux, Kernel 6.15.4

### Additional system information

GPU: NVidia GTX 3060, driver version: `nvidia-open-dkms 575.64-1`

### Printer

N/A

### How to reproduce

1. Launch Orca Slicer (I run `orca-slicer`) from the terminal
2. Click on menu
3. Click on "preferences"
4. Program freezes
5. Program terminates with `SIGSEGV (Address boundary error)`

### Actual results

Program terminates with `SIGSEGV (Address boundary error)`

### Expected results

I should be able to see the preferences dialogue. 

### Project file & Debug log uploads

[debug_Sun_Jun_29_16_01_06_2754717.log](https://github.com/user-attachments/files/20970563/debug_Sun_Jun_29_16_01_06_2754717.log)

### Checklist of files to include

- [x] Log file
- [ ] Project file

### Anything else?

This occurs on both the official AppImage version, the git version in the AUR (`orca-slicer-git`).

I understand the AUR version is not officially supported, but since it tracks the development git branch it may be useful for tracking down this issue.

MathisP75, 2025-06-30:
Same here, I get `segmentation fault (core dumped)  orca-slicer`

lodriguez, 2025-07-18:
I maintain the AUR-git version. Downgrading Pango to 1:1.56.3-1 works as a workaround. I will take a closer look at this later.

ipatch, 2025-07-19:
thanks for tracking this down @lodriguez 🙏 i was in the process of building orca-slicer from source using homebrew deps, but ran into all sorts of issues trying to build wxwidgets with cmake instead autotools, and then ran into several homebrew related issues with cmake, wxwidgets, and homebrew 😫

> installing NanumGothic works, but I guess a more robust fix/workaround would be preferable.

would it not make sense to make `NanumGothic` a required dep for the orca-slicer aur package that you maintain so other wouldn't run into this issue, and then have to search through the github issue tracker?

ipatch, 2025-07-19:
i installed the required font package, but orca-slicer is still crashing when i attempt to open the preferences,

```
╰─λ yay -S NanumGothic
 -> No AUR package found for NanumGothic
 there is nothing to do
╭─capin at rogue in /home/capin/homebrew/Library/Taps/homebrew/homebrew-core (ipatch.archlinux.wxwidgets-cmake ✔)
╰─λ yay -Ss NanumGothic
aur/ttf-nanumgothic_coding 2.5-2 (+25 0.00)
    Nanum series fixed width TrueType fonts
```

```
:: Processing package changes...
(1/1) installing ttf-nanumgothic_coding                                        [--------------------------------------------] 100%
```

```
╰─λ coredumpctl list
TIME                            PID  UID GID SIG     COREFILE EXE                       SIZE
Sat 2025-07-05 18:55:02 CDT  141184 1000 984 SIGABRT present  /usr/bin/orca-slicer-bin 91.9M
Wed 2025-07-09 14:40:12 CDT 3142614 1000 984 SIGSEGV present  /usr/bin/orca-slicer-bin 87.9M
Wed 2025-07-09 14:57:03 CDT 3222762 1000 984 SIGABRT present  /usr/bin/superslicer       14M
Wed 2025-07-09 15:00:21 CDT 3243140 1000 984 SIGABRT present  /usr/bin/superslicer     14.1M
Wed 2025-07-16 13:38:43 CDT 3417746 1000 984 SIGSEGV present  /usr/bin/orca-slicer-bin 91.7M
Wed 2025-07-16 14:18:49 CDT 3558680 1000 984 SIGSEGV present  /usr/bin/orca-slicer-bin 84.6M
Wed 2025-07-16 14:39:53 CDT 3602301 1000 984 SIGSEGV present  /usr/bin/orca-slicer-bin 87.9M
Sat 2025-07-19 14:57:55 CDT 1372359 1000 984 SIGSEGV present  /usr/bin/orca-slicer-bin 90.3M
```