OrcaSlicer/OrcaSlicer #10030: Add Orthographic View Mode and Camera FOV Controls

TinieblasGonzalez, 2025-06-30:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

📌 Feature Request: Add Orthographic View Mode and Camera FOV Controls
Description:
Currently, OrcaSlicer only supports a perspective view in the 3D preview window, which results in noticeable distortion and aberration — especially for large models or when working with wide angles.

Request:
Please add the following camera-related features:

Orthographic View Mode
Allow users to switch between Perspective and Orthographic views, as implemented in SuperSlicer and PrusaSlicer. This would improve the modeling and preview experience by removing lens distortion.

Field of View (FOV) Adjustment
Expose a slider or numeric input (e.g., 20° to 90°) for users to adjust the camera’s field of view, reducing distortion for those who prefer working in perspective mode.

Shortcut or Menu Access
Ideally, include a top-level menu item such as View > Camera, or allow right-click access in the empty 3D space to toggle these options.

Why it matters:
Many users working with large-scale models, mechanical components, or exact spatial placement need undistorted views. The current perspective view often exaggerates dimensions, especially at the edges of the screen, making visual checks less reliable.

Platform:
OrcaSlicer version: [insert your version]
OS: [Windows / macOS / Linux]

### Which printers will be beneficial to this feature?

Others, Klipper, Marlin

### Describe the solution you'd like

Allow users to switch between Perspective and Orthographic views, as implemented in SuperSlicer and PrusaSlicer. This would improve the modeling and preview experience by removing lens distortion.

ianalexis, 2025-07-02:
Orthographic already in orca
[image]

Ataraxia-Mechanica, 2025-07-04:
Weirdly phrased reasons, adding abbreviations unnecessarily, and EM dashes? I think every time someone requests some already existing features with ChatGPT the devs should just delete said feature for funsies