OrcaSlicer/OrcaSlicer #10036: Thin (single line) bridges printed as outer wall and not anchored

AnyOldName3, 2025-07-01:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0 70931e5

### Operating System (OS)

Windows

### OS Version

Windows 10 build 19045

### Additional system information

CPU: Ryzen R9 9900X
RAM: 64 GB DDR5
GPU: Radeon RX 9070 XT

### Printer

Sovol SV08

### How to reproduce

* Download `v6_Cartridge_Straight.stl` from https://github.com/nevermore3d/Nevermore_Micro/tree/master/V6/STL
* Slice it with 0.4mm-ish line width. As far as I can tell, no settings make a difference except setting a line width that makes all bridges at least two lines wide.
* Observe that the single-line bridges are coloured like outer walls instead of bridges, and are completely unanchored.

### Actual results

(sliced with 0.4mm line width)

[image]

The bridges are orange like walls, and begin and end in midair.

### Expected results

(sliced with 0.2mm line width)

[image]

The bridges are blue like bridges, and begin and end on top of walls.

### Project file & Debug log uploads

[v6_Cartridge_Straight.zip](https://github.com/user-attachments/files/20988485/v6_Cartridge_Straight.zip)

[debug_Tue_Jul_01_00_59_43_3324.log](https://github.com/user-attachments/files/20988507/debug_Tue_Jul_01_00_59_43_3324.log)

### Checklist of files to include

- [x] Log file
- [x] Project file

drcowsley, 2025-10-02:
Or am I doing something wrong?

Help please.

Are there trigger words to get triage attention?

AnyOldName3, 2026-10-31:
Still a problem with the 2.5.0 beta 345efec64c7b95a05ce5cf96df0dde66a158835c

[image]