OrcaSlicer/OrcaSlicer #10038: Settings Overrides (Retractions) not working with klipper

slowthisbirddown, 2025-07-01:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0

### Operating System (OS)

Windows

### OS Version

11

### Printer

Ender 3 Pro w/ SKR and Klipper

### How to reproduce

On a filament profile I am able to use the checkbox to override retraction speeds and distances but it doesn't make any changes at all. Printer is still using the settings from the profile.

Using Klipper gcode flavor

Firmware retaction is off

### Actual results

Only prints with the settings from my printer profile. Does not change at all no matter how wild I go with the override settings in the filament profile.

### Expected results

I should see either bad stringing or retractions so high half the print actually prints

### Project file & Debug log uploads

thanks

### Checklist of files to include

- [ ] Log file
- [ ] Project file

slowthisbirddown, 2025-07-11:
Yeah unfortunately this is being ignored anywhere I go.

slowthisbirddown, 2025-07-11:
Do you want to give it a try? I can as well.

infodel, 2025-07-12:
Add these into your OrcaSlicers GCODE, not to the printer.cfg's gcode.  Either in the Printer startup gcode itself or for Filament gcode.  I just saw them automatically adjust within klipper during its print initialization. 

I added these to my Gcode Start Macro on Orcaslicer 

SET_RETRACTION RETRACT_LENGTH=[filament_retraction_length]
SET_RETRACTION RETRACT_SPEED=[filament_retraction_speed]
SET_RETRACTION UNRETRACT_SPEED=[filament_retraction_speed]

I added this to the Filament for PETG to slow it down.
; Filament gcode
M220 S50
; Set Print Speed to 50%

slowthisbirddown, 2025-07-12:
This allows per filament overrides to work? 

So I don't have to put any numbers in the [brackets]?

I can't believe nobody is talking about this.

infodel, 2025-08-15:
This is my Printer start gcode that is specific for filament.  So for PETG i use this.   I can then adjust the advanced settings in the filament options, and they are adjusted on my printer in klipper when the print runs. 

SET_RETRACTION RETRACT_LENGTH=[filament_retraction_length]
SET_RETRACTION RETRACT_SPEED=[filament_retraction_speed]
SET_RETRACTION UNRETRACT_SPEED=[filament_deretraction_speed]
; Filament gcode
M220 S35
; Set Print Speed to 35%

infodel, 2025-08-19:
Were you able to get it to work successfully? Im curious, I dont see a lot of details about it online anywhere.

infodel, 2025-08-20:
Awesome to hear you might have made progress.  Its been really nice to not manually set things like retraction length and speed, and general print speed, for different filaments.