OrcaSlicer/OrcaSlicer #10045: [Feature Request] Z-Index Anti-Aliasing

robgraeber, 2025-07-02:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

Seen this reddit post where someone implemented z-index aliasing for the top surface. If I understand correctly this basically adds variable layer heights for the top layer down to 0.05 mm by dynamically varying the height of extrusions within a single layer. Seems like it could be a killer experimental feature for OrcaSlicer to add and there is already code for it.

https://www.reddit.com/r/BambuLab/comments/1lpfhhw/i_implemented_nonplanar_slicing_for_bambu_studio/
Bambu Studio Fork: https://github.com/adob/BambuStudio-ZAA

### Which printers will be beneficial to this feature?

Others

### Describe the solution you'd like

Implement Z Anti Aliasing Feature based on code from https://github.com/adob/BambuStudio-ZAA

tlhintoq, 2026-02-08:
If its already in a branch of BambuStudio, I would think it shouldn't be a nightmare to merge in.  Right?

vgdh, 2026-02-12:
> I just downloaded the latest nightly build to test it out but I am having trouble finding the setting in the interface. Could you clarify if the ZAA contouring is already integrated into the nightly builds or if I still need to build it manually from your fork for now?

It's not in the orca dev version yet. 
But you can download the build from the pull  request https://github.com/OrcaSlicer/OrcaSlicer/actions/runs/21869111129