OrcaSlicer/OrcaSlicer #10050: Somehow the gui gives me motion sickness

NikoQT, 2025-07-02:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

I just installed the slicer for the first time and somehow I cant look at it for longer than 10 seconds without getting severe motion sickness. I've never had this problem with any other software

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Therefore I think it should be possible to change the font type and maybe size. Other accessibility settings also would be good I guess

moth4017, 2025-07-03:
do you have blue eye?   also try the  dark mode

NikoQT, 2025-07-04:
> do you have blue eye? also try the dark mode

I have green eyes and I already tried the dark mode but it doesn't help.

moth4017, 2025-07-04:
the reason i ask about your eye colour i was in the AV industry for many years and when LED TVs and projects first came out we had a number of customers complaing of  mation sickness  most of them >90%  had blue eyes