OrcaSlicer/OrcaSlicer #10058: wall printing twice as thick

moth4017, 2025-07-03:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0 build 70931e5

### Operating System (OS)

Windows

### OS Version

win10

### Printer

valkyrie ht 

### How to reproduce

in the slicer the wall is shown as 0.6mm yet it prints at 1.2mm , wall on the part is 0.61mm 

### Actual results

wall prints at 1.2mm

[image]

### Expected results

wall to print at 0.6mm

[image]

### Project file & Debug log uploads

[wall issue.zip](https://github.com/user-attachments/files/21047281/wall.issue.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

moth4017, 2025-07-03:
I found the issue its laying down 2 layers at the same height , take a look at the g code e.g at layer 12 height 3.3 but its laying down 2 layers with slightly different extrusion amounts

Ataraxia-Mechanica, 2025-07-04:
[image]
Classic mode isn't really programmed to deal with thin walls like this and is glitching out, try arachne

moth4017, 2025-07-04:
the  line witdh is narrower than  the model wall, and trying prusa  slicer works perfectly

Ataraxia-Mechanica, 2025-07-04:
uhhhh, then I have no clue. Arachne works with you profile on my end tho.

moth4017, 2025-07-05:
i have now tried changing the nozzel size down to 0.4 , and printing line width to 0.55, but still has the issue of printing  2 rounds at the same height .

moth4017, 2025-07-08:
ive also tried super slicer ,  no issue 

[image]

moth4017, 2025-07-19:
tried  the night build , issues  still pressent 

[image]

moth4017, 2025-07-23:
[image]
issue still in this version

moth4017, 2025-07-31:
issue still in this release

[image]

moth4017, 2025-08-04:
issue still in 

[image]

moth4017, 2025-08-09:
issue still there in this version @SoftFever 

[image]

moth4017, 2025-08-09:
reported  by other  user
https://discord.com/channels/1137181739773603922/1389777323162599444

https://discord.com/channels/1137181739773603922/1363262234767261777/1383651547623460934

SoftFever, 2025-08-13:
I can reproduce this bug.

For now, the workaround is either to turn off "Detect thin walls" when using the classic engine, or to use Arachne.

moth4017, 2025-08-13:
detect thin walls is not a solution , and arachne changes the wall thick ness ,  if you  use my project file can you not see the issue?
@MxBrnr  can also repeat the issue

MxBrnr, 2025-08-13:
Yes, I can confirm this exact same thing has happened in several cases with a 0.4 line width as well. One of the main purposes of using the Classic wall generator is to have exact line widths that do not vary, but that gets negated when the slicing engine thinks it needs to print two 0.40mm walls on top of each other for a model with a wall thickness of 0.40mm. I agree that while `Detect thin walls` and switching to Arachne may be considered work arounds, they are not suitable solutions for the underlying problem, especially when that problem does not exist in other Slic3r derivative slicers. I will provide additional project and log files shortly.

SoftFever, 2025-08-17:
> detect thin walls is not a solution , and arachne changes the wall thick ness , if you use my project file can you not see the issue? [@MxBrnr](https://github.com/MxBrnr) can also repeat the issue

I want to clarify the following: to print properly using the Classic Wall Generator for you part, you need to enable Detect Thin Walls. This also applies to PrusaSlicer. The bug I mentioned occurs when printing thin parts without thin wall detection enabled—in this case, walls should not be generated for those parts. Currently, OrcaSlicer generates double paths, which is incorrect. PrusaSlicer, on the other hand, simply does not generate walls at all, which is the expected behavior IMO. 
Regardlessly, if you want to print this part with a 0.6mm nozzle using the Classic Wall Generator, enabling Detect Thin Walls is still necessary.  
I hope this clarifies the issue.

moth4017, 2025-08-31:
Alpha release  night build 
still hase the same issue of printing the same layer twice

[image]

MxBrnr, 2026-01-09:
Another user on Discord has reported this issue happening to them with a single wall print: https://discord.com/channels/1137181739773603922/1458268544369295401/1459047248381083790

mark4017, 2026-02-26:
is there anyone looking at these issues?