OrcaSlicer/OrcaSlicer #10063: Build surface  adding/editing

moth4017, 2025-07-05:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

it would be good to either edit the build surface  names and/or add new ones 

### Which printers will be beneficial to this feature?

Others

### Describe the solution you'd like

it would be good to either edit the build surface  names and/or add new ones

tlhintoq, 2025-07-06:
You already can change the names of the plate and add more.

Is this your first time working with PC software?  That's not meanness... there's a big movement with the younger generation to get away from PC's and work solely in mobile environments like phones and tablets.  So I get it if you're not familiar with the same environments that many of us grew up on.

Using the toolbar at the top you can add what you're calling a "build surface", referred to as a 'plate' in this software.

And that icon next to the plate name is the edit button.

[image]

moth4017, 2025-07-06:
may be this drop down  could have the ability to add modify the  build surface

[image]

MinoWizard, 2025-08-05:
Adding a second into this. With effects plates being a thing, there's a ton of beds that don't fit the pre-set ones in the slicer, so the ability to add more and tie more settings into them would be much appreciated

TexZeTech, 2025-08-12:
So this was an open FR https://github.com/SoftFever/OrcaSlicer/issues/4655
I would definitely like the ability to add custom named plates and or sub groups.

Arya-Sourav, 2026-02-22:
This would be a really good addition for people like me who use multiple texture beds on their printers.