OrcaSlicer/OrcaSlicer #10069: Adjustable Max/min layer heights for "Variable Layer Height"

HipHopAnonymous69, 2025-07-05:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

I've been fiddling with Variable Layer Height, and love the increase in detail and speed it provides! Utilizing default middle values and a 0.4 mm nozzle results in some layers ~0.08 mm in height, leading to stringing issues.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

I'm aware I could modify portions of heights on the right using left/right clicks, but think it'd be simpler and easier to be able to set hard limits to layers height... say 0.3 max, 0.1 min in my situation.

### Additional context

Thank you for your time and effort on an awesome application!

MxBrnr, 2025-07-05:
You mean like this?

[image]

HipHopAnonymous69, 2025-07-06:
> You mean like this?
> 
> <img alt="Image" width="782" height="622" src="https://private-user-images.githubusercontent.com/142743732/462839687-75939caf-e1e5-4461-99a7-2019d32a4699.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTE3NjAxOTMsIm5iZiI6MTc1MTc1OTg5MywicGF0aCI6Ii8xNDI3NDM3MzIvNDYyODM5Njg3LTc1OTM5Y2FmLWUxZTUtNDQ2MS05OWE3LTIwMTlkMzJhNDY5OS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwNzA1JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDcwNVQyMzU4MTNaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1mODFmODAzYWUwMzUxNjk5MDY0OGZiMDNiYTIzOTlkY2M0YzAyMzMzMGI4MGY2ZmRiZTZjNzJkZDU5MTUyNDcxJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.Ge8caytDu9kBMepppUzjOtAVhjokQ1mISVXKxEZOoo0">

lol, exactly like that! Thanks!