OrcaSlicer/OrcaSlicer #10070: wiki support for community contributions

ipatch, 2025-07-05:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

as of right now there's no clear way to submit a PR for the github wiki. i believe it is a feature that github has not implemented unfortunately but has been brought up several times. i think at the very least having users be able to submit changes to the wiki would really help in keeping the calibration section of the wiki current.

there's an interesting writeup on the wiki could support community contributions.

https://www.growingwiththeweb.com/2016/07/enabling-pull-requests-on-github-wikis.html

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

https://www.growingwiththeweb.com/2016/07/enabling-pull-requests-on-github-wikis.html

ipatch, 2025-07-05:
https://github.com/orgs/community/discussions/50163

ipatch, 2025-07-05:
https://github.com/SoftFever/OrcaSlicer/discussions/7440

moonbard, 2025-09-06:
> I think that with the "new" [How to wiki guide](https://github.com/SoftFever/OrcaSlicer/wiki/How-to-wiki) you can find [what](https://github.com/SoftFever/OrcaSlicer/wiki/How-to-wiki#structure) and how to edit it here. Do you think this is enough?

It doesn't say how to actually EDIT the wiki. I wanted to try out contributing by first fixing some typos on https://github.com/SoftFever/OrcaSlicer/wiki/strength_settings_patterns before more involved contributions, but there's no edit button or different pages, nor an option to do a pull/commit. I can see the revision history but no way to interact.

ianalexis, 2025-09-06:
> It doesn't say how to actually EDIT the wiki

It says exactly what and where you have to edit:
`Wiki Structure
Each wiki page is a Markdown file located in the doc directory of the repository. The wiki is organized into sections that cover different areas of the project.`

> an option to do a pull/commit

thats a standard github feature...
If you check the PR list and history you can see all the wiki edits.

Now im working on the Wiki Update 12 #10452 
These updates are usually focused on something specific, but if someone wants to contribute and doesn't know how, they usually leave me comments about errors or improvements to be implemented, and I correct them by adding the user as a co-author of the commit.

> fixing some typos

I generate the patterns automatically with an xlsx file, and it's probably the most tedious process there is. If you want, let me know what changes need to be made and we'll apply them.

moonbard, 2025-09-06:
> > It doesn't say how to actually EDIT the wiki
> 
> It says exactly what and where you have to edit: `Wiki Structure Each wiki page is a Markdown file located in the doc directory of the repository. The wiki is organized into sections that cover different areas of the project.`

What? That's not referencing anything visible. See screenshot below. Github is a repository so that's pretty vague, and there's no docs menu or link.
I last pushed a commit to github like 15 years ago for one class project and have barely touched it since other than comments to help narrow down issues. I swear the layout used to make sense. Is there a comprehensive single man page for all the "standard" features now? Other than copilot.

[image]

>I generate the patterns automatically with an xlsx file, and it's probably the most tedious process there is. If you want, let me know what changes need to be made and we'll apply them.

Ah, now this is in my current skillset. I do almost everything in excel (even when I'm kludging in things that really should go in Access, but that's what VBA and Power Query scripting is for) so I understand it's a very versatile hammer, but not everything is a nail. Why use a spreadsheet for expository text and embedded images? I maintain a digital catalog for cloud software so it's somewhat similar in goal.

As for the edits themselves, it's not important - the point was shaking off the rust but I see that's not the problem. I'm going to go back to only making comments. I'm probably the only one who noticed aesthetics is misspelled 'Esthetics' with mid sentence capitalization anyway.

ianalexis, 2025-09-06:
> What? That's not referencing anything visible. See screenshot below. Github is a repository so that's pretty vague, and there's no docs menu or link.

Just because you may not be familiar with how GitHub works doesn’t mean the documentation isn’t there. The repository clearly contains Markdown files inside the DOC folder, which is where the wiki content is written and maintained.

> I swear the layout used to make sense. Is there a comprehensive single man page for all the "standard" features now? Other than copilot.

The wiki structure hasn’t changed. The homepage is still the same, but it has been expanded to include all the available options and processes. There’s also an index for navigation, and each section is organized to match the Orca GUI. This was done intentionally so that every Orca item links directly to its corresponding wiki entry.
I’ll be honest, this kind of feedback hits hard. I’ve been working (alongside many other contributors) for months, dedicating hours every week to improving the wiki. So when I read that or unrelated comments about Copilot, it’s discouraging.

> Why use a spreadsheet for expository text and embedded images?

If you check the guide and the notes on the page where Excel is mentioned, you’ll see that it’s not just about text and images. The spreadsheets are used for calculations timing, material usage, strength ratios, etc. and they automatically generate summaries that integrate with the wiki.

That said, the wiki is not meant to teach GitHub basics like cloning a repo, editing Markdown, or submitting pull requests. Its purpose is to maintain order and document the main concepts.

If you’d like more personalized help, I strongly recommend joining the community on Discord. People regularly ask for support there, and many of us are more than willing to guide even through things like compiling the project.

To wrap up: I hope this explanation clears things up. I don’t plan to go into this much detail again, since I believe the above covers your concerns. Please keep in mind that there are real people dedicating their free time to this project, and it’s important to differentiate between constructive feedback and criticism based solely on lack of familiarity.

> aesthetics is misspelled 'Esthetics'

Excellent, will change it ASAP.