OrcaSlicer/OrcaSlicer #10072: Strange seams shown in vase mode

RF47, 2025-07-06:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.1-dev 

### Operating System (OS)

Windows

### OS Version

Windoows 10

### Additional system information

Ryzen 9 5900X
64 GB ram
Rtx 3080Ti

### Printer

Ender 3

### How to reproduce

Make a cylinder Diameter 42mm height 120mm, active vase mode and  show seams. (3mf file attached)

### Actual results

[image]

### Expected results

The bug appears to be prior to commit c6b8664c8e276413445ed25d3a8957101143c006

### Project file & Debug log uploads

[bug seam.zip](https://github.com/user-attachments/files/21084656/bug.seam.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

ianalexis, 2025-07-07:
In my Ender 3, this seam points to ORIGIN coordinante system.
"Fixing" the mesh model will remove the seam until i move it around the plate.

ianalexis, 2025-07-07:
From what I saw, it detects and displays the seam but does not generate retraction gcode or anything.
Maybe not showing seam in vase mode is the solution although maybe it is a bit of hardcode and there is some other logic to apply.

RF47, 2025-07-09:
i found something:

[image]

RF47, 2025-07-12:
> This is not the ideal solution IMO. I changed that to make sure any gcode in change layer/change filament etc. that forgot to move the z back to the correct layer height will be fixed automatically when start printing the model. Without that the nozzle could hit the model, print in mid air or even scratch the heat bed. I know this is more of an user error, but we don't want things like [#8387](https://github.com/SoftFever/OrcaSlicer/issues/8387) to happen again.
> 
> The extra G1 code does not affect the actual print, it just been visualized as a seam somehow, which should be fixed instead.

and what do you think of #10103?

RF47, 2025-07-13:
fixed in #10135