OrcaSlicer/OrcaSlicer #10078: Cosmic Desktop - empty 3D viewer window

jarekdudzinski, 2025-07-06:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0, 2.3.1-dev

### Operating System (OS)

Linux

### OS Version

PopOS 24.04 Cosmic Desktop

### Additional system information

After launching, the 3D preview window where you arrange models on build plate is empty, black. After switching tabs it's white.

NVidia 5060 Ti 16GB RAM, clean OS - just installed fresh system, updated.

### Printer

BambuLabs A1

### How to reproduce

1. install & update clean PopOS 24.04 Cosmic DE (nvidia)

2. fix libraries
sudo ln -s /usr/lib/x86_64-linux-gnu/libwebkit2gtk-4.1.so.0 /usr/lib/x86_64-linux-gnu/libwebkit2gtk-4.0.so.37
sudo ln -s /usr/lib/x86_64-linux-gnu/libjavascriptcoregtk-4.1.so.0  /usr/lib/x86_64-linux-gnu/libjavascriptcoregtk-4.0.so.18

3. start Orca

### Actual results

After launching, the 3D preview window where you arrange models on build plate is empty, black. After switching tabs it's white.

### Expected results

Seeing what I'm doing in 3D viewport? ;)

### Project file & Debug log uploads

no need

### Checklist of files to include

- [ ] Log file
- [ ] Project file

PinkCandyMS, 2025-07-08:
Arch linux hyprland desktop same issue, rtx 4060 gpu propietery drivers. In my case 3d view doesnt get black but when resizing the window some parts goes black. tried using QT_QPA_PLATFORM=xcb but it still refuse to work

jurassicjordan, 2025-07-18:
is there a fix? Im having same issue on Archlinux with KDE plasma 6, rtx 3060 on both X11 and Wayland

EtheriousNight, 2025-07-21:
There is a fix coming for Bambu Studio. I don't know if it gets also ported to Orca Slicer 😅

ArKay74, 2025-07-26:
No rendering in Pop!_OS 22.04 under Wayland (RTX 5090) either.

ArKay74, 2025-07-26:
FWIW, when I use x11 instead of a Wayland session (through the desktop selection on login), it works just fine. On Wayland it's also rendering through x11 (when I force it to use Wayland through Flatseal, it doesn't even start).

ArKay74, 2025-07-27:
Yeah, that was probably a different problem.

PinkCandyDev, 2025-07-29:
Nvm it broke again after some update xd. And i have to use it on x11. I love arch linux

ArKay74, 2025-08-12:
Well, decided to give a nightly a try and what should I say? It works now on Wayland for me.

So no more running an extra x11-Server for me. :)

Maybe give a nightly a try?

xMxMiller, 2025-08-16:
When I install the flatpak, the termial gives me a warning saying this, the branch 46 of org.gnome.Platform is no longer supported.

I'm using nobara 42 with KDE Plasma on wayland.

closed 2025-11-23 by github-actions[bot]