OrcaSlicer/OrcaSlicer #10079: Elegoo Centauri Carbon timelapse not working

nitrokryptic, 2025-07-06:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0

### Operating System (OS)

Windows

### OS Version

11

### Printer

Elegoo Centauri Carbon

### How to reproduce

1. slice an stl
2. Click send, print, or export
3. check "timelapse" if uploading over wifi and turn on the timelapse button if doing manual transfer
4. It's sent to the printer by manual or over wifi 
5. it starts printing and you notice the issue

### Actual results

You get no timelapse after the print is finished. I suspect it is caused by the printer reading no layers, as the layer count stays at zero the whole print. And I need timelapses as a creator and to diagnose issues mid-print. happens with all files.

### Expected results

The timelapse should be exported into the timelapse folder on the printer and be ready for transfer post-print.

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

Just this happening is annoying, and I wonder if I can get it fixed in an update in the near future.

digitalresistor, 2025-07-13:
This was fixed in: https://github.com/SoftFever/OrcaSlicer/pull/9630

digitalresistor, 2025-07-18:
> Issues still present on latest nightly 2.3.1-dev build [e44ec1f](https://github.com/SoftFever/OrcaSlicer/commit/e44ec1f27b362b3eaa72fb44d59fea1d81f05c0f) for MacOS

I've found that OrcaSlicer creates a copy of the machine in `~/Library/Application Support/OrcaSlicer/system/Elegoo` when you customize it (for example by adding the IP address to connect to for the printer). This means the profile is not loaded from the new instance of OrcaSlicer when creating the printer again from disk. So older copies of the printer won't get auto updated when you update to the latest version of OrcaSlicer (including the dev builds). The new file is correctly included in the application bundle.

I couldn't find a way to force Orca to upgrade the G-Code for the machine, other than removing the entire folder and starting from scratch.

axiopaladin, 2025-07-31:
> I've found that OrcaSlicer creates a copy of the machine in `~/Library/Application Support/OrcaSlicer/system/Elegoo` when you customize it (for example by adding the IP address to connect to for the [...]

The equivalent is at `~/.config/OrcaSlicer/` on Linux systems, for anyone else who comes here trying to figure this out later.