OrcaSlicer/OrcaSlicer #10080: Wipe moves happen outside of their object tags

johyphenel, 2025-07-06:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0

### Operating System (OS)

macOS

### OS Version

All

### Printer

Voron Trident

### How to reproduce

The issue in https://github.com/SoftFever/OrcaSlicer/issues/979 is still present in the current build of Orca Slicer. 

### Actual results

Purge tower generates wipe way outside the bounds of the purge tower, which can then be out of bounds for the build plate and cause the print to fail. 

[image]

### Expected results

Wipes should be within the bounds of the object being printed. 

### Project file & Debug log uploads

. 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

johyphenel, 2025-07-06:
Amusingly, the wipe is more likely to be out of bounds if you position the purge tower on the left side of the plate (see screenshot in first post) than if you have it on the right: 

[image]

johyphenel, 2025-07-08:
Maybe has to do with how close the parts are to the front edge of the plate as well..?

[image]

VigilantGardener, 2025-08-12:
I am also experiencing this issue, very specifically the direction of the wipe seems to be effected by which build plate the prime tower is on.  In all cases the direction of the wipe will be towards the back side or left side of build plate 1.

I can't attach a 3mf, but here is a video showing the test setup and demonstrating the problem.
https://youtu.be/4rBuEZ3Nius

You can observe this behavior by slicing all plates, and then in the preview go to "line type" and enable visibility of wipes.  The behavior will appear as follows
Plate 1 - all wipes move toward 12 o'clock or 9 o'clock
Plate 2 - some wipes point to 9 o'clock others are angled towards 12 o'clock on plate 1
Plate 3 - some wipes point to 9 o'clock others are angled towards 12 o'clock on plate 1
Plate 4 - some wipes point to 12 o'clock others are angled towards 9 o'clock on plate 1
Plate 5 - all wipes are angled towards plate 1, either 12 o'clock or 9 o'clock
Plate 6 - all wipes are angled towards plate 1, either 12 o'clock or 9 o'clock

VigilantGardener, 2025-08-12:
@johyphenel   Can you try the experiment as I laid out in my update?  Are you experiencing the same build plate dependent behavior?

johyphenel, 2025-11-11:
Issue still present

noobydp, 2025-12-24:
Issue still present