OrcaSlicer/OrcaSlicer #10090: Crashes after startup on latest Nvidia

CommanderT3706, 2025-07-07:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

orca-slicer 2.3.1.c6b8664c8e276413445e-2 (Built from AUR)

### Operating System (OS)

Linux

### OS Version

Arch Linux

### Additional system information

CPU: Intel i5 12400F
RAM: 32GB DDR4
GPU: Nvidia RTX 3050

### Printer

Creality K1 SE

### How to reproduce

1. Ensure `nvidia-open` 575.64.03-2 is being installed
2. Start Orca Slicer with `orca-slicer`
3. Start Orca Slicer with `orca-slicer-bin`

### Actual results

`orca-slicer` starts, but crashes shortly after
`orca-slicer-bin` starts, but renders nothing in the main viewport

### Expected results

Orca should start normally

### Project file & Debug log uploads

`orca-slicer` reports no useful logs other than some GTK errors
`orca-slicer-bin` reports `Failed to create GBM buffer of size 2560x1300: Invalid argument`

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

This issue began after using the `nvidia-open 575.64.03-2` driver.
I suspect it has something to do with the driver being updated, impacting Vulkan/OpenGL in some way

larchypaws, 2025-07-08:
I experienced the same issue following upgrade to nVidia 575.64.03. This affected Orca 2.3.0 stable appimage and flatpak, and 2.3.1 dev flatpak and aur.

A full OS reinstall did not rectify the issue.

I was able to rectify by downgrading CacyOS's nVidia driver version to 575.57.08 using the command:

sudo downgrade nvidia-dkms nvidia-utils lib32-nvidia-utils lib32-opencl-nvidia libva-nvidia-driver opencl-nvidia

Following this all versions of Orca are working again normally.

Kryvonis, 2025-07-09:
Exactly the same for me, tried it from AppImage and from orca-slicer-bin, crashes with this error logs. 
I'm so glad I'm not alone 🙏🏻

```
(orca-slicer:1831047): Gtk-CRITICAL **: 19:55:47.120: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed
(orca-slicer:1831047): Gtk-CRITICAL **: 19:55:47.122: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:1831047): Gtk-CRITICAL **: 19:55:47.231: gtk_window_resize: assertion 'height > 0' failed

** (orca-slicer:1831047): CRITICAL **: 19:55:47.327: Cannot register URI scheme wxfs more than once

1. lo    : 127.0.0.1
2. wlp13s0: 192.168.1.213

No such file or directory

(orca-slicer:1831047): Gtk-CRITICAL **: 19:55:48.525: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed
```

larchypaws, 2025-07-09:
On the orcaslicer subreddit someone posted a workaround of installing Virtual GL and then running vglrun orca-slicer. This does seem to work, albeit with the home page not rendering, but everything else does. Hope this may help in the meantime.

tlhintoq, 2025-07-10:
Just asking - don't shoot me...
But isn't the newer 576.80 available to you for that card? My 3080 uses it and the 3050 is only 7 months older.

[image]

larchypaws, 2025-07-10:
> Just asking - don't shoot me... But isn't the newer 576.80 available to you for that card? My 3080 uses it and the 3050 is only 7 months older.
> <img alt="Image" width="1040" height="404" src="https://private-user-images.githubusercontent.com/11351506/464492604-443be828-d6c4-441b-b61d-db979c6f389c.PNG?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTIxMzY0NTgsIm5iZiI6MTc1MjEzNjE1OCwicGF0aCI6Ii8xMTM1MTUwNi80NjQ0OTI2MDQtNDQzYmU4MjgtZDZjNC00NDFiLWI2MWQtZGI5NzljNmYzODljLlBORz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA3MTAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwNzEwVDA4MjkxOFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTgyZjZiNzZjYzE4OTJhYWRkODg3YjAxMWM2ZjJiOTM3MGMxZGNmZWJkNDZmYjU4ZGJiN2MzYjc1ZDRhNDYwMGQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.j0bkav_LiHtT1qtWEfCC0XJsCWR3g01DSMopOqgtJpQ">

Not on the Linux distribution I am using.

CommanderT3706, 2025-07-10:
> Just asking - don't shoot me... But isn't the newer 576.80 available to you for that card? My 3080 uses it and the 3050 is only 7 months older.
> <img alt="Image" width="1040" height="404" [...]

I think that's a different driver to what we have on linux

JellyJones, 2025-07-10:
It seems to be an issue with zink, as removing it from the environment variables causes it to run in software mode perfectly fine.

tlhintoq, 2025-07-11:
> It seems to be an issue with zink, as removing it from the environment variables causes it to run in software mode perfectly fine.

Really?  They didn't keep version numbers in sync across OSes?  What a f'ing mess that is.  At least keep the major digits in sync to avoid these kinds of confusions.

DeathofaStar, 2025-07-17:
same issue

degid, 2025-07-20:
Same issue, but I think the problem is not in the nvidia driver. The crash occurs in wayland, but it works fine in X11.
CPU: AMD Ryzen 5 5600X
RAM: 128GB DDR4
GPU: NVIDIA GeForce RTX 5070 Ti
```
> pacman -Q | grep nvidia
lib32-nvidia-utils 575.64.03-1
linux-firmware-nvidia 20250708-1
nvidia-open 575.64.03-4
nvidia-settings 575.64-1
nvidia-utils 575.64.03-1
```
Problematic configuration:
```
> plasmashell --version
plasmashell 6.4.3
> echo $XDG_SESSION_TYPE
wayland
```

Working configuration:
```
> echo $XDG_SESSION_TYPE
x11
```

[image]

johnny-bit, 2025-07-25:
Darn... I've tested on my "main" PC and I've got the same issue.

```
$ echo $XDG_SESSION_TYPE
x11
$ nvidia-smi --version
NVIDIA-SMI version  : 575.64.05
NVML version        : 575.64
DRIVER version      : 575.64.05
CUDA Version        : 12.9

```

Running "normally" causes
```
Failed to create GBM buffer of size 820x621: Invalid argument
```

Using workaround env from https://github.com/SoftFever/OrcaSlicer/issues/6433#issuecomment-3037220708
allows to run Orca with just
```
nouveau: driver missing
```
warning, but everything seems to work.

This leads me to believe this might be driver issue since I remember Orca working on my main PC not that long ago...

Can anyone test with drivers 570 (stable) and see whether the issue still occurs? I'll try to test over the weekend
If it's nvidia driver fault it would be great to post feedback to https://forums.developer.nvidia.com/c/gpu-graphics/linux/148 so that nvidia is made aware of it.

bmbeverst, 2025-07-29:
In case GitHub goes down again, the magic command is:
`env -u WAYLAND_DISPLAY XDG_SESSION_TYPE=x11 WEBKIT_FORCE_COMPOSITING_MODE=1 WEBKIT_DISABLE_COMPOSITING_MODE=1 GBM_BACKEND=dri /usr/bin/orca-slicer`

Or for AppImage:
`env -u WAYLAND_DISPLAY XDG_SESSION_TYPE=x11 WEBKIT_FORCE_COMPOSITING_MODE=1 WEBKIT_DISABLE_COMPOSITING_MODE=1 GBM_BACKEND=dri  ~/Applications/OrcaSlicer_Linux_*`

From this [thread](https://github.com/SoftFever/OrcaSlicer/issues/6433#issuecomment-3037220708).

Lucretia, 2025-07-29:
> In case GitHub goes down again, the magic command is: `env -u WAYLAND_DISPLAY XDG_SESSION_TYPE=x11 GBM_BACKEND=dri /usr/bin/orca-slicer`

This worked for me.

Kryvonis, 2025-08-11:
> In case GitHub goes down again, the magic command is: `env -u WAYLAND_DISPLAY XDG_SESSION_TYPE=x11 GBM_BACKEND=dri /usr/bin/orca-slicer`
> 
> Or for AppImage: `env -u WAYLAND_DISPLAY XDG_SESSION_TYPE=x11 GBM_BACKEND=dri ~/Applications/OrcaSlicer_Linux_*`
> 
> From this [thread](https://github.com/SoftFever/OrcaSlicer/issues/6433#issuecomment-3037295054) as [@johnny-bit](https://github.com/johnny-bit) pointed out!

Almost worked. It started an orca but when I tried to open file it crashed

Kryvonis, 2025-08-20:
I attempted the fix [feat: Add automatic graphics backend configuration system #10313](https://github.com/SoftFever/OrcaSlicer/pull/10313), and it worked to some extent. 
However, it only functions properly when I drag and drop files. Additionally, certain features, such as exporting to STL, are causing errors. 
It seems that anything related to modal windows is causing the application to crash.

[image]

klylabs, 2025-08-25:
> `__EGL_VENDOR_LIBRARY_FILENAMES=/usr/share/glvnd/egl_vendor.d/50_mesa.json orca-slicer-bin` works for me, but it uses software rendering rather than hardware rendering
> 
> EDIT: It broke today

updating to the latest nvidia 580.76.05 should fix the issue entirely (as long as you use the appimage version if you're on arch)