OrcaSlicer/OrcaSlicer #10093: Screen Reader labels for controls

vablus, 2025-07-07:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

None of the controls for Orca slicer have labels for screen readers to read. I have tested this out using several screen readers as I'm a blind user of the program and nothing reads. This includes buttons, combo boxes, tree-views, panels, sliders, etc. All of the controls for every window on Orca slicer need screen reader labels so that I and a lot of people can navigate the program. Also, the arrow keys and tab key do not work for navigating theUI at all and unfortunately, it requires a mouse to use which is both very unusual and rather inconvenient since blind folks do not use mice regularly.  Because of these issues, Orca Slicer is genuinely unusable for a subset of the 3d printing community and believe me, I've spent hours trying to find even the most basic options. 

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

All windows on Orca slicer need labels for their controls so that screen readers to be able to read what is happening. for example, instead of hearing, "button, button, button", you should ideally hear, "Filament type, Filament temperature, Printing speed", or whatever it says that sighted people caan just look at. I don't know what the windows say because screen readers such as JAWS, NVDA, Narrator, Orka, and VOiceOver do not read any of the controls including when things are changed and the values. 

### Describe alternatives you've considered

Unfortunately, I have to use Prusa slicer to get anything done because it is currently the best, although rather imperfect, solution. I have long wanted to use Orca slicer because I hear it is the best. It isn't the best for me or other blind makers.

Aitch-ko, 2026-11-01:
In case it helps: #16788 was merged on 30 October and should be in the current nightly builds.

What it fixes:

- screen readers announce Orca's custom buttons and checkboxes as buttons
- their labels and tooltips are read out
- tested with NVDA and Narrator on Windows, and VoiceOver on macOS

What it does not cover yet: keyboard navigation and the other control types asked for here.

@vablus, if you try a nightly with your screen reader, a short note here on what works now and what does not would help the next step.

vablus, 2026-11-02:
Hello there.

So, I tested out the nightly build of Orka Slicer. I love how the homepage is navigable. I managed to create an Orka cloud account for the first time and I'm so glad that most of the stuff on the page works.
However, when I load in a file, there's no way to switch from the canvas to any of the other pages so that I can navigate the other windows. Ideally, there should be a keyboard shortcut, perhaps ctrl+alt+1 through 6, to switch the windows that you had mentioned are now labeled.
Also, this may be a little more difficult, but I made a rudimentary canvas grid audio navigator that is supposed to show the model so that a blind person can at least hear it. It doesn't really work, but it's an idea that I worked on using Claude and I wanted to put it out there for someone who might have some more ideas on how to make the actual grid accessible just as much as the now labeled screens.
Please do what you can to fix these issues, particularly the lack of keyboard shortcuts for transitioning to the windows so that we can navigate the labeled windows which I very much appreciate.

daiverd, 2026-11-09:
@vablus thanks for testing. In the current nightly, Ctrl+Tab should switch to the next main tab (Home, Prepare, Preview and so on); it's listed under Help > Keyboard Shortcuts. If it doesn't work from the 3D view, that's worth reporting here.

My PR [#15290](https://github.com/OrcaSlicer/OrcaSlicer/pull/15290) adds the rest of the navigation: F6 and Shift+F6 move between the top bar, the tab strip and the current page, and the arrow keys move along the tabs. It's waiting for review.

I'd like to see your grid navigator; could you share it as a repo or a gist? Making the canvas usable without sight is the hardest part of this, and an audio approach is worth exploring even if it doesn't work yet.