OrcaSlicer/OrcaSlicer #10105: Flow rate calibration printing top surface as concentric

james633, 2025-07-09:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.1-dev / eba08bf nightly build

### Operating System (OS)

Windows

### OS Version

Windows 10

### Printer

Pruse CORE One

### How to reproduce

Load the included Prusa CORE One profile and Polymaker PA612-CF or any default filament profile, and start a flow rate calibration.

### Actual results

Flow rate calibration bricks print the top surface concentrically, regardless of top surface settings saying Monotonic.

[image]

### Expected results

I installed the nightly build 2.3.1-dev / eba08bf to get a Prusa CORE One printer profile. All of the flow calibrations are printing the top surface pattern as what appears to be concentric. The top surface setting is is Monotonic as you can see in the screenshot, but all surfaces end up slicing (and printing) showing concentric like this. Is this intentional? No matter what top surface pattern I set to there is no change.

### Project file & Debug log uploads

[Orca YOLO Flow Calibration.zip](https://github.com/user-attachments/files/21150209/Orca.YOLO.Flow.Calibration.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

### Anything else?

Here's a shot of a print 

[image]

MxBrnr, 2025-07-09:
This is intentional. The test has been redesigned to use this new pattern. There are new instructions on how to use it too: https://github.com/SoftFever/OrcaSlicer/discussions/9609#discussioncomment-13395354

james633, 2025-07-09:
Ah thanks somehow this didn’t come up in my searches. I guess maybe looking for concentric threw them off.

strayr, 2025-07-10:
>  [#9609 (comment)](https://github.com/SoftFever/OrcaSlicer/discussions/9609#discussioncomment-13395354)

Hey @MxBrnr it looks like if you link to your own comments it doesn't work for everyone else due to your privacy settings. This link works.

https://github.com/SoftFever/OrcaSlicer/discussions/9609

MxBrnr, 2025-07-11:
> Hey [@MxBrnr](https://github.com/MxBrnr) it looks like if you link to your own comments it doesn't work for everyone else due to your privacy settings. This link works. [...]

Interesting. Thanks for the heads up. The idea was to link straight to the instructions instead of making people scroll through the whole discussion. Does this one work? https://github.com/SoftFever/OrcaSlicer/discussions/9609#discussioncomment-13395354