OrcaSlicer/OrcaSlicer #10108: Crashes if 2 or more users use the slicer

im-vmn, 2025-07-10:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0

### Operating System (OS)

Linux

### OS Version

Linux 6.15.4-arch2-1 #1 SMP PREEMPT_DYNAMIC x86_64 GNU/Linux

### Printer

Elegoo Neptune 4

### How to reproduce

1. Open slicer and slice something
2. Switch to another user (or even log in with VNC)
3. Ty to slice something under the second user

### Actual results

[image]

### Expected results

My model will be sliced

### Project file & Debug log uploads

[debug_Thu_Jul_10_10_31_51_6001.log](https://github.com/user-attachments/files/21155626/debug_Thu_Jul_10_10_31_51_6001.log)

### Checklist of files to include

- [x] Log file
- [ ] Project file

### Anything else?

1. Remove **/tmp/orcaslicer_model** directory
2. Restart the slicer
3. Everything is fine for the current user

jrha, 2025-12-19:
This appears to be the same problem as #5969, the underlying issue being that the slicer expects to have exclusive use of a shared temporary directory.

`pam-tmpdir` provides another workaround

im-vmn, 2025-12-19:
Thank you, I hoped to avoid pam-tmpdir, solution with a script from #5969 seems more elegant, since it's not always possible to install additional packages.

closed 2025-12-19 by im-vmn

amalon, 2025-12-19:
this would still appear to be categorically a bug in orcaslicer though. it should create a uniquely named tmp directory to avoid clashes with other users.