OrcaSlicer/OrcaSlicer #10116: Different PA

twyt-egg, 2025-07-10:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

PA slows down the print, in some places unnecessarily much

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

it would be nice to be able to set different PA values for different line types like in ultimaker cura. 
For example 0.3 for outer walls and 0.12 for infill.
Because one PA value for everything slows down my printer too much (it takes nearly 30% longer) and with this setting you could set the best PA value for outer walls where you can see the differenz and set it lower or of for places like infill that you cant see

RF47, 2025-07-21:
> > since if there is overextrusion in the infill, it is not a problem.
> 
> It will be a problem if you have nearly solid infill though.

In the worst case it is just as well to have pressure advance disabled, it is not a big deal.
Also, in that case you can always leave the checkbox unchecked, it is a similar option to disabling retraction for infill.

RF47, 2025-07-21:
This option can also be beneficial for tpu printing, as reducing extruder retraction movements reduces the possibility of heat creep and clogging.