OrcaSlicer/OrcaSlicer #10126: Orcaslicer Flatpak added to Flathub?

Scout339, 2025-07-11:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

Looks like there isn't an actual discussion around having the flatpak uploaded to Flathub, where everyone goes to install and update the app. I've been using the 2.3 beta flatpak, but it would be amazing if I didn't have to constantly come to the Github to update my only app on my Fedora device that is manually installed.

### Which printers will be beneficial to this feature?

Others

### Describe the solution you'd like

Add Flatpak's to Flathub when they are updated on Github.

Pandapip1, 2025-11-03:
Seems to be a dupe of https://github.com/SoftFever/OrcaSlicer/issues/3949

Scout339, 2025-12-25:
Apparently I'm closing due to it being a dupe, even though it's not resolved.

closed 2025-12-25 by Scout339