OrcaSlicer/OrcaSlicer #10140: Lost camera feed

loudpa, 2025-07-14:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.1 Dev Ubuntu2404

### Operating System (OS)

Linux

### OS Version

Linux Mint 22.1

### Printer

Elegoo Centauri Carbon

### How to reproduce

When I started to use Orca day 1 I would send a print to the print and as expected the video would appear showing the printing process. This worked perfectly over the course of  two days.
After that after sending a print I no longer get a video feed. It states "Connection error" when one tries to refresh.
In the WiFi setting the connection test states working.
If I use a browser and the printer's IP I receive video.

### Actual results

The video feed is blank, if one refreshes it states connection error.

### Expected results

Video of printing should show.

### Project file & Debug log uploads

### Checklist of files to include

- [x] Log file
- [x] Project file

### Anything else?

Keep getting failed trying to attach files requested

closed 2025-07-14 by loudpa

kaibagley, 2025-10-04:
Can this be reopened? I'm having the blank camera issue too, but changing and resetting the device UI IP doesn't solve the issue.

dsptech11-prog, 2026-08-19:
Similar issue. When the camera doesn't work I have to fill out the Device UI field with "Centauri-Carbon" . After that the camera feed appears in the Device tab. Next time I try to print another item the camera feed is out again. I have to go back and clear the Device UI field to get it working again. 
It's every other item I print that does this. Printing the same item an additional time is fine. 
Printer is a Centauri Carbon CC1. Running Orca slicer 2.4.2