OrcaSlicer/OrcaSlicer #10153: OrcaSlicer crashes when trying to open prefferences

BlathanAevon, 2025-07-16:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0

### Operating System (OS)

Linux

### OS Version

Arch Linux  [v6.15.6-arch1-1] 

### Additional system information

Arch Linux  [x86_64]                                       
20W0008VUS  ThinkPad T14 Gen 2i                            
Linux  [v6.15.6-arch1-1]                                   
11th Gen Intel(R) Core(TM) i7-1185G7  [C:8] [4.80 GHz]     
Iris Xe Graphics  [C:96] [1.35 GHz] [Integrated]           
dwl (Wayland)  ( running with xorg-xwayland)                               

### Printer

SOVOL SV06 ACE

### How to reproduce

1. Open app
2. Try to open preferences via gui or via Ctrl+P

### Actual results

App hangs, crashes then, restart doesn't help, re-installing doesn't help, nightly build doesn't help, AppImage doesn't help, AUR versions (orca-slicer-bin, orca-slicer-unstable-bin, orca-slicer) doesn't help

### Expected results

Preferences window to open

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/21257527/log.zip)

### Checklist of files to include

- [x] Log file
- [ ] Project file

ipatch, 2025-07-16:
can confirm this as well, running arch linux.

BlathanAevon, 2025-07-19:
Yeah, indeed. 

The fix is: You need to install ttf-nanum from AUR. ttf-nanumgothic_coding does not include NanumGothic. ttf-nanum has all the nanum fonts.

closed 2025-07-19 by BlathanAevon