OrcaSlicer/OrcaSlicer #10160: Multiple plates do not save when saving as a 3MF

ALTC-AmLearningToCode, 2025-07-17:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

fd0d716

### Operating System (OS)

Windows

### OS Version

Windows 11

### Additional system information

Ryzen 7 5700G
64 GB 3600 MT/s ram
Radeon 7800 XT

### Printer

Ender 3 V3 KE (Custom profiles, not using system profiles)

### How to reproduce

1. Create new project
2. Add another plate, so there is two plates
3. add one primitive object to each plate
4. save project (as 3MF)
5. close and reopen project

### Actual results

After reopening the project, there is one plate, and the two different objects are placed in a way that they are both on the edge of the plate.

[image]

### Expected results

Both plates should be there, with objects placed on each plate as they were when the project was saved. shown image is what would be expected, but this is before saving and reopening the project.

[image]

### Project file & Debug log uploads

[TEMPaddtoissue.zip](https://github.com/user-attachments/files/21279822/TEMPaddtoissue.zip)

[debug_Wed_Jul_16_21_04_58_12064.log.zip](https://github.com/user-attachments/files/21279831/debug_Wed_Jul_16_21_04_58_12064.log.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

witc-git, 2025-09-06:
Same issue. It also doesn't change modifications to global or per object presets. For example, my profile has 3 walls, 3 top shells and 3 bottom shells with brims disabled. I have a project with 3 plates of objects, 2 plates have painted brims and 2 walls modified in the global tab, the third plate has print by object enabled on the plate and 2 objects with 4 walls modified in the object tab.
When saving the 3mf file, none of those modifications are saved within that project.
(I know I can save it as a preset, but these settings are only used for this one project in which I sometime modify text for other prints, hence needing a saved project that retains all the settings I am likely to forget to change)

Scarjit, 2025-09-07:
This might to be a duplicate of or at least strongly related to https://github.com/SoftFever/OrcaSlicer/issues/9288 ?