OrcaSlicer/OrcaSlicer #10179: Ability to increase infill wall/ line count

dandontknow1, 2025-07-19:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

An old feature from Cura that would be useful for certain projects - the ability to increase the amount of infill walls/ line count

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Implement the ability to increase the amount of infill wall/ line count.

RF47, 2025-07-20:
Fill multiline was already implemented in nightly build.
https://github.com/SoftFever/OrcaSlicer/pull/9432

dandontknow1, 2025-07-20:
Great, thanks! Will wait for the stable release.