OrcaSlicer/OrcaSlicer #10181: Recomended  nozzel tempeture limit to low

moth4017, 2025-07-20:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0

### Operating System (OS)

Windows

### OS Version

win 10 pro

### Printer

valkyrie  HT pro

### How to reproduce

if you set the recomened  nozzle  tempreture to the manufactures  specification  in this setting you get an error message 

### Actual results

[image]

### Expected results

not to put a cap on the nozzle temperature as low as 300C

### Project file & Debug log uploads

[temperror0.zip](https://github.com/user-attachments/files/21332197/temperror0.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

moth4017, 2025-07-20:
has nothing to do with the printer , there is a limit in the orca softwasre , you can stil set the extruder  to 400C 

[image]

ianalexis, 2025-08-26:
Is not a bug but just a warning. But i have an idea to improve this.
/assign-me

moth4017, 2025-10-20:
the warning is still popping up in ver 2.3.1

[image]