OrcaSlicer/OrcaSlicer #10184: Add setting for bridge anchor/overlap distance, or default to inner wall

ADKMechETECH, 2025-07-20:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

When printing flexibles (and even when printing other materials), bridges are only anchored to the walls by a single extrusion width.  See attached image.   Even when printing with support material.  With flexibles in particular, this is not sufficient and in high stress parts, this area tears out

[image]

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

It would be really nice to have a setting to control how much overlap there is on a "bottom layer" of the bridge.  Either % of total wall thickness, or number of shells with a limit of the total number of shells for the layer. 

If that is not feasible, the extrusion should default to go to the inside of the *inner wall* and not just the outer wall.  

### Describe alternatives you've considered

1.  Increase flow ratio - not ideal for a variety of reasons.  
2. nothing, because I can't seem to find any other settings that change this behavior.

ADKMechETECH, 2025-07-20:
I did not, assuming it would cause other issues.  The amount I would have to increase it would be excessive and even still I don't think it will solve the issue I have.  Realistically, the bridge should extend to the inner wall.  I can not think of any reason it should stop at inside of the outer wall by default.

moth4017, 2025-07-20:
i think the default in orca slicer is a 15% over lap

moth4017, 2025-07-20:
that would normally be enough , if your under extruding you may get issues

ADKMechETECH, 2025-08-11:
> Still no solution? This is a problem for me too. Prusaslicer and Bambu studio don't seem to have this problem..

As far as I know, no. I don't understand how this is a problem in debut slicer also since Orca slicer is based on it.?  I haven't used bamboo Slicer in a very long time since I like orca better. What is the behavior in bamboo slicer?