OrcaSlicer/OrcaSlicer #10187: On fresh install cannot skip Setup Wizard

melinerunen, 2025-07-20:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0

### Operating System (OS)

macOS

### OS Version

Sequoia 15.5 (24F74)

### Additional system information

On a fresh install, there is no way to skip the setup wizard. I've all the profiles and printers stored in the Bambu Studio, and I'm not able to skip the setup wizard.

[image]

### Printer

Ender 3 pro base with Manta M5p board(and other stuff)

### How to reproduce

Clean uninstall Orca slicer by deleting the following folders/files:

[image]

Install a fresh copy of Orca Slicer.

### Actual results

Opening the app forces to use the setup wizard, with no way to skip it.

### Expected results

Allow me to connect to the Bambu Studio account and sync the saved printer, profiles and filaments from my account.

### Project file & Debug log uploads

No log file required. (if you need any, ill provide)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

Love the app!!!

MxBrnr, 2025-07-20:
This would be a feature/enhancement request, not a bug report.

melinerunen, 2025-07-20:
I didn't realize I created bug! Can you please re-label it? Thanks!