OrcaSlicer/OrcaSlicer #10188: Line widths >1mm are reported as 'invalid values in the 3mf' (even if they are <5x nozzle diameter)

malventano, 2025-07-21:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

Current nightly

### Operating System (OS)

Windows

### OS Version

11

### Additional system information

N/A

### Printer

X1C

### How to reproduce

Set any line widths >1mm.
Save the 3mf.
Open the 3mf.
Note the following error as the file is opened:

### Actual results

[image]

### Expected results

This error should not appear unless the values present are >5x the nozzle diameter, which is the maximum [set](https://github.com/SoftFever/OrcaSlicer/issues/1745#issuecomment-1673394709) by @SoftFever in https://github.com/SoftFever/OrcaSlicer/issues/1745. Setting widths >5x nozzle diameter throws a different error (as expected). For consistency, these should both match the same 5x diameter limit so that the user does not see an error when opening a project that was error free when saved.

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [ ] Log file
- [ ] Project file

moth4017, 2025-07-21:
[test0.zip](https://github.com/user-attachments/files/21339973/test0.zip   here is my 3mf file

malventano, 2025-07-21:
> i tried it in the nightly build and it works ok

I tried opening your test0 and got a warning about jerk values?

Here's mine on the newest nightly:
[image]

3mf (all X1C defaults, only line widths changed):
[test-1.5mm line width.zip](https://github.com/user-attachments/files/21350331/test-1.5mm.line.width.zip)

moth4017, 2025-07-21:
oening your file i got the same error as you i fixed it by  increasing the layer height

moth4017, 2025-07-21:
if you just change the fist layer height it seems to fix it

[image]

malventano, 2025-07-22:
The layer height is in the allowed range.

malventano, 2025-07-22:
> here is the funny thing, chage it to 0.3  and slice  then change it back to 0.2 and slice and the
> res no error

Yes, the error occurs when opening the file.

moth4017, 2025-07-23:
sorry, i was saying that from all the nozzel diamerter i tried it was 2.5 x diameter or 5 x  the radius of the nozzel , ive never seen a ref to  5 times diameter . the limit in the software is 2.5 times .

malventano, 2025-07-23:
Please read the initial post fully. I linked the issue where Orca sets this to 5x.

malventano, 2025-07-24:
> thats correct in "vase mode" it is 5 X

Yes, and the error still appears when opening a vase mode .3mf with line width less than 5x nozzle width.

moth4017, 2025-07-24:
Hi , if i set it to vase mode first  and then set the line  width  to x 5 its all ok

[image]

moth4017, 2025-07-24:
correct the nozzel dia is in the 3mf file but the error  checking happens when you open the file ,  line width  x5 only applies in vase mode and when you slice in vase mode the error goes away. the max line width  in  non vase mode is  x 2.5,  

i only checked the nozzel dia as i thourght 2mm dia  as you said earlier  was rather large  and could have been a wornout nozzle .

malventano, 2025-07-24:
The 5x limit applies regardless of vase mode. Models with line widths 5x nozzle diameter slice just fine regardless of mode (as expected). Again, the issue is that the error appears while opening a 3mf with line widths in the allowable range set by the lead Orca Slicer developer.

Unsure where the disconnect is, but you do not seem to understand that you are not helping to resolve the issue. Saying it's not an issue or being confused about vase mode or the physical properties of a particular nozzle does not solve anything.

moth4017, 2025-07-29:
Sorry it's taken a while to get back I had a reply from the orca dev they said no larger than 2.5 x dia

malventano, 2025-07-29:
> Sorry it's taken a while to get back I had a reply from the orca dev they said no larger than 2.5 x dia

Yes, so this is still a bug as it should not throw an error message when the value is in the allowed range.
Which dev did you ask, and did you also mention this issue?

moth4017, 2025-07-29:
It was from softfever, lodged bugs have to go here  I just wanted the actual value.

malventano, 2025-07-29:
> It was from softfever, lodged bugs have to go here I just wanted the actual value.

At the top of this thread is a link to @SoftFever specifically stating the limit is 5x nozzle diameter. Why are you wasting his time with a question that [he already answered two years ago](https://github.com/SoftFever/OrcaSlicer/issues/1745#issuecomment-1673394709)?