OrcaSlicer/OrcaSlicer #10189: Request: Add OrcaSlicer to Flathub

nilbest, 2025-07-21:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

I’d love a streamlined, official, and easy‑to‑install OrcaSlicer for Linux.  
- When I switch to a new PC, setting up OrcaSlicer takes time.  
- Beginners can be confused by multiple download formats: local Flatpaks vs. AppImages.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Publish an official Flatpak on Flathub so any Linux user can:
- Install OrcaSlicer with one command: `flatpak install flathub org.orcaslicer.OrcaSlicer`  
- Receive automatic updates  
- Avoid hunting through GitHub releases

### Additional context

### Why Flatpak?  
- Cross‑distribution sandboxed package  
- Centralized updates via Flathub  
- Reduces confusion between AppImage and Flatpak versions

### References  
- Flathub packaging guide: https://docs.flatpak.org/en/latest/first-build.html

nilbest, 2025-10-18:
> Im just a regular user, and it would be very convenient for me to install OrcaSlicer via Flathub. I hope it becomes available there someday.

@topssspot Try the way user @NanashiTheNameless https://github.com/SoftFever/OrcaSlicer/issues/10189#issuecomment-3099935123 described. It works then like it should and gets added to your flatpaks and also the system interface.

You can also download the Flatpak version via your browser from the official webpage [https://www.orcaslicer.com/](url) and then install it via `flatpak install --user ./Downloads/<name of the flatpak file>` 

Example:
`flatpak install --user ./Downloads/OrcaSlicer-Linux-flatpak_V2.3.1_x86_64.flatpak`

After that, it should work the same as added via the official flatpakrepo.

nilbest, 2025-10-18:
> > > Im just a regular user, and it would be very convenient for me to install OrcaSlicer via Flathub. I hope it becomes available there someday.
> > 
> > 
> > [@topssspot](https://github.com/topssspot) Try the way user [@NanashiTheNameless](https://github.com/NanashiTheNameless) [#10189 (comment)](https://github.com/SoftFever/OrcaSlicer/issues/10189#issuecomment-3099935123) described. It works then like it should and gets added to your flatpaks and also the system interface.
> > You can also download the Flatpak version via your browser from the official webpage [https://www.orcaslicer.com/](url) and then install it via `flatpak install --user ./Downloads/<name of the flatpak file>`
> > Example: `flatpak install --user ./Downloads/OrcaSlicer-Linux-flatpak_V2.3.1_x86_64.flatpak`
> > After that, it should work the same as added via the official flatpakrepo.
> 
> but yeah I dont think OrcaSlicer will ever be listed on FlatHub for similar reasons that FlatHub won't list that Ventoy -> Flatpak fork ("proprietary stubs/binaries" in reference to the BambuLab network plugin.)

Yeah, that's true, we may be able to close this Request. I think a short how-to would help a lot of people to get started after the potential switch from Windows to Linux in the future. It seems to me that I'm not the only one who bumped into this issue, and it took me some time to find out how it should be done. (I know it is more a user problem of flatpak)
But as I said, a short How-to may help a lot of potential noobies on this part.

NanashiTheNameless, 2025-10-18:
> Yeah, that's true, we may be able to close this Request. I think a short how-to would help a lot of people to get started after the potential switch from Windows to Linux in the future. It seems to [...]

100% agreed more documentation is good!

NanashiTheNameless, 2025-10-18:
> > but yeah I dont think OrcaSlicer will ever be listed on FlatHub for similar reasons that FlatHub won't list that Ventoy -> Flatpak fork ("proprietary stubs/binaries" in reference to the BambuLab network plugin.)
> 
> I don’t think that’s an issue here... Not sure what happened with the Ventoy fork, but here’s what the Flathub docs have to say on the matter: https://docs.flathub.org/docs/for-app-authors/requirements#license.
> 
> > All content hosted on Flathub must allow legal redistribution, and the license must be correctly specified in the app's [MetaInfo file](https://docs.flathub.org/docs/for-app-authors/metainfo-guidelines#license).
> > Non-redistributable sources must use [extra-data](https://docs.flatpak.org/en/latest/module-sources.html#extra-data) source type. Any part of the application such as the name or icon must not violate any trademarks and must be distinct.
> 
> AFAIK the blobs are downloaded at run time so Flathub shouldn’t have an issue hosting OrcaSlicer without any modification. I’d guess Bambu Studio wouldn’t be on Flathub either if there was no way to use the network plugin.

Good point, I may be mistaken

coryrc, 2025-12-25:
Please close as dupe.