OrcaSlicer/OrcaSlicer #10199: Wrong temperature in preview (0), but in GCODE is correct

Marek-Dvorny, 2025-07-22:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0

### Operating System (OS)

Windows

### OS Version

Windows 11

### Additional system information

CPU: AMD Ryzen 7000 SERIES, GPU: AMD Radeon

### Printer

Voron 2.4 R2 Pro with ERCF V2 12 channels

### How to reproduce

Open the attached 3MF; click "Slice all" or "Slice plate"; select temperature preview - there is written 0, but in gcode it's correct.

### Actual results

The all temperature is marked 0, but in GCODE it is correct.

### Expected results

The temperature should be 250 in this case.

### Project file & Debug log uploads

[Stray Kids STAY.zip](https://github.com/user-attachments/files/21366305/Stray.Kids.STAY.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

### Anything else?

[image]

jusii, 2025-07-22:
Found this too yesterday, but if you have varible temps like with temperature tower, it displays temps just right.

Semphora, 2025-09-21:
I'm experiencing a version of this. if I open a new prject and add any primitive the displayed temperature states 150C. The Gcode is correct. As an aside, when slicing the temp tower, the correct temperatures are displayed.

MxBrnr, 2026-04-21:
This is still an issue in both v2.3.2 "stable," and v2.4.0-dev build 2e61d9e.
It only happen when a single temperature is used for the whole print, and can further confirm that it is only a preview issue; G-code does use the correct temperature.

christopherbaxter, 2026-05-20:
Hello team, specifically, first layer temp is previewing as 0 (I set my first layer to be 251 and the rest to be 250) first layer is showing as 0deg in the preview, while the rest are showing as 250deg. Hopefully this helps the team to find the fault.

im-vmn, 2026-11-08:
The same issue. 

1. Orca version 2.5.0
2. Linux my-host 6.12.115+deb13.1-amd64 # 1 SMP PREEMPT_DYNAMIC Debian 6.12.115-2 (2026-11-01) x86_64 GNU/Linux
3. Cube primitive
4. Qidi Q2 printer profile

[image]